SuperDARN/rst #4: Removing object and other compiler generated files?

ksterne, 2017-02-03:
I think this was a to-do for me a while ago, but should we remove all of the object files, binaries and other files that are created during the compilation process?  I see the ~/bin/ directory is still removed here, but object files are still up.

egthomas, 2017-02-03:
I did this in a commit on the feature/grid2 branch (https://github.com/SuperDARN/rst/commit/9ed327b335e74c302bef3e0472b3cbc8f077b9ec) but it's kind of buried with a few other gridding-specific commits.  It's probably not something that should have been rolled out together.

egthomas, 2017-02-06:
Just cleaned up the *.a, *.so, and *.o files (139f267ea0e654364154d348b3244f2a821a7916).  Is there anything else that should be removed?

egthomas, 2017-02-06:
Ok, after today's round of commits the develop branch should no longer be tracking any files that are created during the compilation process.  After a fresh clone/download of the rst and compilation the only files that show up from a "git status" are the base.bash and idl.bash files (which must be modified to set some user-dependent environment variables).

ksterne, 2017-02-13:
Looks like we're a lot better here. So, closing this out.

closed 2017-02-13 by ksterne