SuperDARN/rst #5: Twofsound multiple 'channel' data not being processed correctly (example 20161117 CDN data)

ksterne, 2017-02-03:
This is the issue from the old repo (https://github.com/vtsuperdarn/VTRST3.5/issues/25) moved over.  Here I've copy and pasted Kevin Krieger's original post.

For some reason, RST3.5 doesn't process fitacf files correctly for the twofsound data from twofsound.1.04 where the channel parameter is changed to '1' or '2' depending upon which frequency is being transmitted.

Below are plots from davitpy that used fitacfs generated at usask with stk, they seem to have the data plotted. Also attached are plots from Dieter's IDL plotting code written using stk libs, which don't seem to have plotted correctly, for completeness.

[image]
[image]
[image]
[image]

pasha-ponomarenko, 2017-02-07:
Any news here? I am struggling to find time for looking closer into this issue with three conferences looming in the next 5-6 weeks. In order to be able to have a detailed look I would need to replicate the badlag determination in IDL so that I can easily track what is going on. I will try...

ksterne, 2017-02-07:
Haven't found the time here either, spent most of the time transitioning over to this repo.  I think the latest thing I need to take a look at is why the receiver blanking isn't being marked.  I've tried taking a look at the cross range interference issue as well, but haven't had a chance to look at it.

ksterne, 2017-05-06:
Believe this has now been fixed with #57.  Do you believe that's true @pasha-ponomarenko?

pasha-ponomarenko, 2017-05-06:
I hope so! 
;-)

egthomas, 2017-05-06:
Ok, closing.

closed 2017-05-06 by egthomas

egthomas, 2017-05-16:
Is it also safe to remove the `bugfix-usask_freq_chan` branch?

ksterne, 2017-07-12:
Sorry to be slow, but it seems good to remove this branch so I've done so.