SuperDARN/rst #6: vt idl library

egthomas, 2017-02-07:
So there currently is a folder of IDL routines in codebase/superdarn/src.idl/vt/ where it looks like AJ uploaded some of his old software.  As several of these routines are dependent on common blocks and functions native to Davit (Lasse Clausen's IDL package, not the newer python software), it doesn't seem appropriate to keep all of this stuff in the RST.  I guess my question is whether anyone has an objection to removing this directory (and the related output_plots directory)?  These IDL routines would still be archived in VTRST3.5 on the vtsuperdarn github account if someone really wanted to access them.

sshepherd, 2017-02-07:
I would remove them.

egthomas, 2017-02-07:
Similarly, do we need to keep the compare_files.py script?  This looks like the only piece of python software in the RST and was again added by AJ.

ksterne, 2017-02-07:
Yea, I'm not what these files are for.  It seems as though they are IDL versions of some of the fitacf processing routines?  Like you say, they are being backed up in the older VTRST3.5 repo, so it should be safe to remove them.  I definitely agree with removing the `compare_files.py` code.  That is a neat short script on comparing the file lengths of two fitacf files.  But this could easily be reproduced if needed.

pasha-ponomarenko, 2017-02-08:
Kevin, I think compare_files.py has been developed by AJ when we satrted refactoring FITACF in August 2014. We wanted to be sure that consecutive modifications ot the package produce the same result.

egthomas, 2017-02-13:
Closing this issue as it was addressed by pull request #9 as well.

closed 2017-02-13 by egthomas