SuperDARN/rst #8: Development schedule

egthomas, 2017-02-07:
I'm going to quote @ksterne from the same issue he had previously opened on the VTRST3.5 repository:

> Even though there isn't a lot of active development going on with this repository, would there be any opposition of mirroring the branch merging schedule of davitpy? It's been quite a while since develop was merged into master. And though the develop branch is set to be the default, it may be good to keep master up-to-date as new users/devs will assume master is the best version.

This seems like the kind of thing that would help remove any confusion about which branch is the "definitive" version to clone.  That way our develop branch could actually function as a beta version as we begin implementing upgrades to make_fit/make_grid/AACGM/MLT.

ksterne, 2017-02-07:
I think around this, we should also move the default git branch here from develop to master.  I've tentatively set an 0.1 release branch creation for April 26th and then the 0.1 release (merge develop into master) on May 31st.

For reference, this is assuming a commonly used devleopment model of: http://nvie.com/img/git-model@2x.png

egthomas, 2017-02-07:
Shouldn't we maintain consistency with the previous RST version numbering?  ie if we start with anything less than 3.5, eventually we would have different RST distributions with the same version number.

ksterne, 2017-02-07:
That makes sense, I hadn't thought about the potential confusion there.  So then I guess we're on 3.6?  I think I've seen ros 3.6, but not rst.

pasha-ponomarenko, 2017-02-08:
I second calling it RST3.6.

egthomas, 2017-02-09:
OK, so it sounds like we should aim to have a v3.6 release ready in time for the SD workshop in June.  Would it also be useful to tag the point where the repository officially diverged from VTRST3.5 (58842cb) as v3.5?  I'm not very familiar with git's tagging system but I assume there is a sensible way to do so for historical purposes.

ksterne, 2017-02-10:
It is probably a good idea to mark the commit that this repo and the VTRST3.5 repo shared as the 3.5 release.  That will give a good common starting point as we continue to develop.  I haven't done the release creation too much if you'd like to give it a go @egthomas.

egthomas, 2017-02-11:
I think v3.5 has now been properly tagged - someone more familiar with git should let me know if I messed it up.

ksterne, 2017-02-11:
Yep I think you've got it.  I'll close this now as it seems as though we're agreed on dates and naming here.

closed 2017-02-14 by egthomas