SuperDARN/rst #10: Merging and/or removing stale branches

egthomas, 2017-02-09:
It looks like the following branches carried over from the VTRST3.5 repository may be stale / abandoned based on their commit history:

- cs10
- make_fit2
- bugfix-IDL_time_conversion
- bugfix-fitstat
- feature/refactorfitacf

The cs10 branch is where AJ began implementing Ellen's CS10 statistical model into Map Potential; I intend to resurrect that effort at some point over the next year.  The make_fit2 branch looks like an older attempt by AJ to refactor fitacf and has possibly superseded by the feature/refactorfitacf branch ( @pasha-ponomarenko ?).  @ksterne , can you speak to the bugfix-IDL_time_conversion and bugfix-fitstat branches?  It looks like the bugfix-IDL_time_conversion made a simple change that would be easy to merge but the resolution of the effort on the bugfix-fitstat branch was not quite so obvious to me.

ksterne, 2017-02-13:
The IDL time conversion branch is an old issue that Rob Fear brought up where he was seeing some issues with changing dates or years.  As you noted, this goes back to almost a year ago, or maybe to just before the 2016 workshop.  Time has been a little bit of a blur since then.  In all my free time, I'll try to get on that more since I think we had it mostly figured out.

The fitstat bugfix branch was an effort to get the fitstat function to work again.  I can't say I remember as much on this, but I think at one time running fitstat didn't yield the correct or expected results, if it worked at all.  For comparison, I know rawstat does work though.

egthomas, 2017-03-14:
@pasha-ponomarenko and @kkotyk - are the pasha-ponomarenko-patch-1, simtodevel, or simtomaster branches still necessary or can they be deleted?  It seems like these were addressed by pull request #19.

kkotyk, 2017-03-14:
Pretty sure simtodevel and simtomaster can be. I had thought those were deleted already to be honest so I would say go ahead!

ksterne, 2017-03-23:
I've deleted the bugfix-fitstat branch here as I see @kkotyk merged this in last summer.  A week or so after that was a blur for quite a while so my apologies on not following up or remember.

egthomas, 2017-03-30:
@ksterne it looks like the change made in the bugfix-IDL_time_conversion branch has been implemented by @sshepherd when he updated astalg.pro for this AACGM_v2 libraries:

https://github.com/SuperDARN/rst/blob/develop/codebase/analysis/src.idl/lib/astalg.2.0/astalg.pro#L76-L94

Do you think that branch is safe to remove as well?

egthomas, 2017-04-05:
@ksterne - not sure if the above comment slipped by you last week

ksterne, 2017-04-05:
@egthomas without relooking at this and relying on my poor memory, yes I agree that I think the time conversion was fixed up.  Sorry I think I figured this out a few weeks ago and noted the fix to Rob Fear who had brought it to my attention.  So, I think we're good to remove that branch as well.  So, issue done?

egthomas, 2017-04-05:
There are still the pasha-ponomarenko-patch-1, feature/refactorfitacf, and make_fit2 branches that either @pasha-ponomarenko or @kkotyk can hopefully shed some light on.  My guess is that the pasha-ponomarenko-patch-1 branch can be removed after the cross-range interference pull request was approved (#19).  It also seems like the feature/refactorfitacf and make_fit2 branches may be irrelevant now that the refactored version of fitacf has been developed over on the SuperDARN Canada github repository.

kkotyk, 2017-04-05:
the feature/refactor branch has two versions of fitacf added. Version 2.6 is the same as 2.5, although I cleaned up a lot of code(names, refactoring, etc). 2.7 builds on 2.6 to add some bugfixes that pasha outlines on the DAWG page(I think). He can comment on those changes. 3.0 is a complete rework of the project featuring several improvements. 3.0 can be merged into RST once the community is ready to accept the results.

pasha-ponomarenko, 2017-04-05:
I believe we need to preserve both fitacf2.6 and 2.7 one way or another -- they are the most advanced versions of fitacf2 to date, and fitacf2 is still the "official" package.

egthomas, 2017-04-05:
Ok, I think I understand now.  So does that mean the feature/refactor branch will eventually be merged into develop?  And does that mean AJ's make_fit2 branch is no longer necessary with the newer fitacf.3.0?  (note that it will still exist in the VTRST repository)

pasha-ponomarenko, 2017-04-05:
I am not sure what make_fit2 branch is/was about.

kkotyk, 2017-04-05:
@pasha-ponomarenko should feature/refactor be merged into develop?

egthomas, 2017-04-05:
From the commit names on the make_fit2 branch, it looks like AJ started creating a "new cleaner fitacf" in Dec 2013 (just before he graduated and left).  Unless there are any strong objections I'm going to go ahead and remove this branch since it appears to be superseded by fitacf.2.6 / 2.7 / 3.0.  It can always be copied over from the VTRST repository in the future if necessary.

@pasha-ponomarenko - do you still need the pasha-ponomarenko-patch-1 branch?  the only new commit there has since been added by Keith

egthomas, 2017-05-01:
I think we've now taken care of pretty much all of these, and since no one has spoken up in defense of the make_fit2 branch (which again can be found on the VTRST repository) over the last 3.5 weeks I'm going to go ahead and remove it and close this issue.

closed 2017-05-01 by egthomas