SuperDARN/rst #11: Merging corrections to the simulator from the refactor branch

pasha-ponomarenko, 2017-02-16:
As I am still pretty much a dummy with github, can anyone help me to merge these changes form the refactor branch into the main obne? 
https://github.com/SuperDARN/rst/commit/5035d30d9dc09413eb2e5e2dfb36369b627da8f4
These changes have been tested awhile ago, but I have not managed to marge them properly...

egthomas, 2017-02-16:
Is this the only commit from the refactor branch that you intended to merge?  Or will others be merged eventually as well?

ksterne, 2017-02-16:
It seems as though @kkotyk, made the most of the commits in the `feature/refactorfitacf` branch.  Would he be able to lead the pull request?  If this pull request isn't merging correctly due to my commit a week or two ago, we can forgo that with what is in the currently develop branch.

kkotyk, 2017-02-16:
Can we merge that whole branch Pasha or just that one commit?

pasha-ponomarenko, 2017-02-16:
Just this one please.

Cheers,
Pasha

---- kkotyk wrote ----

Can we merge that whole branch Pasha or just that one commit?

kkotyk, 2017-02-17:
I need permissions to be able to create branches and perform pushes.

egthomas, 2017-02-17:
You should have a pending invitation to join the SuperDARN organization with write privileges for this repository.

kkotyk, 2017-02-17:
I double checked my notifications and emails and I don't see anything. Can you resend it at all?

egthomas, 2017-03-02:
Closing this issue as the pull request (#19) was approved and merged.

closed 2017-03-02 by egthomas