SuperDARN/rst #12: ROS software

egthomas, 2017-02-16:
Since I haven't heard a response from the OS-WG chair (Mike) I figured I'd go ahead and throw this out there.  There is currently a fair amount of ROS software located in the codebase/ and usr/ directories of the RST.  From what I can gather, this is all MSI-specific, and thus is not relevant to the majority of SuperDARN users or radar operators.  Furthermore, Kevin and Simon both appear to have backups of the ROS for their own MSI radars stored on github (I imagine UAF has something similar).

My question is this:  should the ROS software also be removed from RST?  It is a very straightforward task and already implemented, tested, and successfully compiled on the ros_cleanup branch. I believe this is one of 2, possibly 3, remaining steps in reducing the RST to a package we are able to fully support.  (the other steps I can foresee are removal of the rpkg component [implemented and tested on the pkg_cleanup branch], and removal of the plotting software)

ksterne, 2017-02-16:
I'll speak up for Mike since I don't think he's on Github at all.  I know this week he's been slammed with something.  We've seen him in the office some, but sometimes with the door shut.  I can bug him in the flesh sometime next week if you'd like.

I am in favor of separating out the ROS code from this repo as I think we should strive for RST to be a data processing packaging.  There's a few parts of code that could be called data reading software that I think would also be good to keep.  Software that runs radar electronics seems like a completely different goal or use.  It is true that individual groups have their own backups for radar code.  How dependent RST code is on ROS code, I'm not sure.  I'd suspect not very much.  I think there's probably more dependency of ROS code on RST code for fit-level data generated at a radar.

In addition to being a software package we can better support, it will make the RST package leaner and cleaner.

egthomas, 2017-02-16:
I had contacted Mike via email but that doesn't surprise me if he's busy with other things.  When you say data reading software do you mean something that was in one of the os/ directories of the RST?  If so, could you point out the particular binaries/libraries which were deleted from the ros_cleanup branch?  Otherwise I'll go ahead and create a merge request.

closed 2017-03-06 by egthomas