SuperDARN/rst #26: Docker files

kkotyk, 2017-03-03:
I'm working on creating Docker(https://www.docker.com/what-docker) files so that we can have container images to be able to quickly test installs and stuff in a virtualized environment. I'm just wondering where to keep the files. I think it would be a good idea to start a new repo so that we can create and modify images as we see fit.

kkotyk, 2017-03-03:
pinging @egthomas. I've got Docker files made up for both Ubuntu and OpenSUSE. They are working quite nicely.

egthomas, 2017-03-03:
I don't see any problem with putting these in a separate repository here on the superdarn organization.  Do you have permissions to create a new repository?  (I need to do some more reading to understand how these docker files work)

kkotyk, 2017-03-03:
@egthomas I dont have permissions to create a repo.

egthomas, 2017-03-03:
ok that should be fixed, let me know if you still don't have permission

egthomas, 2017-03-07:
Do we need any additional docker files?  I could try creating one more for Mint.

kkotyk, 2017-03-07:
Up to you I guess. If the Mint install process isn't too complicated then maybe its a good idea.

egthomas, 2017-05-01:
Probably not going to get to that Mint docker file after all so I think we're done here.  @kkotyk - agreed we can close this issue?

kkotyk, 2017-05-01:
Agreed.

closed 2017-05-01 by egthomas