SuperDARN/rst #30: Integration of cmpfit causing small differences in fitex files

ksterne, 2017-03-07:
As found in #16, somewhere between the VTRST3.5 tag and that pull request, a change in the codebase appeared to be causing a differences in the p_l and w_l variables in the result of `make_fitex2 -new`.  Only looking at a few of the first differences, it appears as though the change in the values is very small, but every difference, even within a 2 hour file, was not categorized.  Even if all of the differences are small, this make may comparing datasets tricky if the contents of the files are different.

pasha-ponomarenko, 2017-03-07:
I looked at the code (fitacfex2.c, lines 415-438), and it looks like estimating p_l and w_l does not involve MPFIT. Instead, it is based on a least-square procedure replicated from the Numerical Recipes, nrfit.c.

egthomas, 2017-03-07:
To test this issue, I followed these steps on a Debian 8.7.1 virtual machine:

1. clone develop branch of VTRST3.5 repository from vtsuperdarn account
2. compile VTRST3.5 (painfully)
3. execute `make_fitex2 -new 20161012.1601.00.bks.rawacf > 20161012.1601.00.bks.fitex.vtrst`
4. clone develop branch of rst repository from superdarn account
5. compile RST
6. execute `make_fitex2 -new 20161012.1601.00.bks.rawacf > 20161012.1601.00.bks.fitex.rst`
7. execute `dmapdump -d 20161012.1601.00.bks.fitex.vtrst > 20161012.1601.00.bks.dump.vtrst`
8. execute `dmapdump -d 20161012.1601.00.bks.fitex.rst > 20161012.1601.00.bks.dump.rst`
9. execute `diff 20161012.1601.00.bks.dump.vtrst 20161012.1601.00.bks.dump.rst | tee output_file.txt`

Examining the contents of output_file.txt, the only differences between the VTRST3.5 output and RST output are the string "origin.time" values.  This is the same result I observed when comparing the output of make_fitex2 of the develop and cmpfit_cleanup branches of this repository.

So in summary, I am unable to replicate this issue on either an Ubuntu or Debian system.

ksterne, 2017-03-07:
@egthomas, that is certainly interesting....I'll have to take a look at some more testing to see what is going on here.  As @pasha-ponomarenko noted, it could be that it's a different version of cmpfit, or did you use the version that is with the VTRST3.5?  I think I remember we had ended up including that package within that repo...but maybe I'd left it out for copyright issues.

egthomas, 2017-03-07:
The above test used the version of cmpfit (1.2) which is contained in both the VTRST3.5 and RST repositories.  We haven't integrated the newer version of cmpfit (1.3a) that Pasha identified into any of the RST repositories yet.

pasha-ponomarenko, 2017-03-08:
After a  closer look I identified that in fitacfex2.c no routines using mpfit are utilised. While the lmfit.h is called in the header, the only procedures used from that package are following:
calc_phi0 
setup_fblk
lm_noise_stat
calc_err
All of them are AJ's replicas of the respective FITACF routines, and neither of them uses the mpfit library.
Therefore, so the source of the discrepancy observed by Kevin should be somewhere else.

egthomas, 2017-03-30:
Is this still a problem @ksterne ?  Have you had a chance to do any more testing?

ksterne, 2017-04-05:
I have not had anytime to do more testing.  I would say this is still an issue.

egthomas, 2017-10-06:
So, it's been 6 months - any movement on this issue?  From looking at the respective histories of the make_fitex2 binary and fitacfex2 library I still don't see how any changes to the RST would impact the values in fitex2 files.

ksterne, 2017-10-06:
Hey @egthomas, I haven't been able to come back to this issue here.  I certain want to since you noted that you didn't see the problem.  Maybe it'll get on my to-do list next week...

egthomas, 2018-03-06:
So it's been a full year and more than 500 commits later with no evidence of this issue.  At the time I compared VTRST3.5 and pre-RST4.0 output on two different Linux distributions and found no differences, and @pasha-ponomarenko confirmed that make_fitex2 and fitacfex2 do not depend on the cmpfit library.  I move that we close this issue.

ksterne, 2018-03-09:
I'll sound like a broken record and I just haven't gotten to testing this yet.  I'd still like to investigate this though.  I was definitely seeing differences in fit-level data between the two repos.  It could be something local to VT though.

aburrell, 2018-03-09:
Fair enough.  We should put a log warning when the routine is run, so people who may use it know that there are potential issues.

ksterne, 2018-03-09:
Eh, I'm not sure a warning message is warranted yet.  I'm ok with keeping this as an issue until I can revisit things and see where I'm at.

aburrell, 2018-03-09:
Ok, you’ve looked at the routine the most so it’s your call.

sshepherd, 2018-03-09:
I think that since this issue has sat idle for over a year, it should be
closed. Kevin can do an internal check if and when he has time, but keeping
an issue open for a year without anyone looking at it just seems a bit
absurd.

egthomas, 2018-06-11:
I think we agreed during the 22 March 2018 telecon this issue would be closed if no action was taken by the time of the SD workshop, so closing.

closed 2018-06-11 by egthomas

ksterne, 2018-06-11:
Yep, absolutely agree.  It doesn't seem to be a big issue, so I guess we can keep in in the closed part for future reference.