SuperDARN/rst #32: Licensing of RST

sshepherd, 2017-03-09:
In the current RST, every .c and .h file has the following statement:

/*
 LICENSE AND DISCLAIMER
 
 Copyright (c) 2012 The Johns Hopkins University/Applied Physics Laboratory
 
 This file is part of the Radar Software Toolkit (RST).
 
 RST is free software: you can redistribute it and/or modify
 it under the terms of the GNU Lesser General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 any later version.
 
 RST is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU Lesser General Public License for more details.
 
 You should have received a copy of the GNU Lesser General Public License
 along with RST.  If not, see <http://www.gnu.org/licenses/>.
 
 
 
*/

The IDL .pro files have something similar. It would be better to replace this comment block with something like:

/*
 * see LICENSE.txt
 */
and I have a script that will do that. The question is what should be in the LICENSE.txt file?

Any thoughts?

sshepherd, 2017-03-09:
Didn't mean for that dot to appear before "see LICENSE.txt" was an asterix.

ksterne, 2017-03-09:
I think the licensing got a little funny as time went on.  I've somewhat thought RST was still a JHUAPL product, but I see here that this isn't tied down (GNU LGPL).  I have a license file for rst.2.11, but it's not at all like this so that probably won't help us.

Are all of the license headers the same?  I would think then that license content as it is now goes into the LICENSE.txt file.

sshepherd, 2017-03-09:
I guess the question is whether it is still copyrighted by JHU/APL. Especially once we start adding new capabilities. Mike or Keith?

ksterne, 2017-03-09:
I think @mikeruo is outside of the SuperDARN team, so we'll need to add him or e-mail him direct for an answer.

sshepherd, 2017-03-09:
Can you ask Mike? I seem to recall that there was something worked out when
the RST _left_ APL. I am wondering if we can just replace the Copyright (c)
2012 JHU/APL with something like Originally developed at JHU/APL and then
the rest about the public license, etc.

kkotyk, 2017-03-09:
This is all definitely before my time here, but I wonder if it's possible to get ownership transferred to us. Who at APL would be responsible for transferring that ownership? I think if thats possible you could leave their copyright line, but just add one right after saying Copyright 2017 SuperDARN, etc, etc.

pasha-ponomarenko, 2017-03-10:
Is anyone left at APL who can update us on this issue? Ethan?

egthomas, 2017-06-13:
Just want to point out that I've found a few instances of misspellings of the word "License" as "Licence" where a previous find-and-replace script appears to have failed, e.g.:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.dlm/iqdlm.1.6/iqidl.c#L7

ksterne, 2017-06-13:
Misspelled? Those might be fightin' words for our Canadian colleagues. 

--------------------
....thumbed from my phone; auto-correct may have improved this message...

----- Reply message -----

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.dlm/iqdlm.1.6/iqidl.c#L7

egthomas, 2017-06-23:
Really digging in the weeds here, but there was some discussion of modifying the APL copyright in the RST license ~10 years ago (see slide 8):

https://web.archive.org/web/20170223192212/http://www.tiger.latrobe.edu.au/superdarn2008/procCD/presentations/10010.pdf

kkotyk, 2017-06-23:
@egthomas  you should win a spoon award for most obscure superdarn webpages found.

kkotyk, 2017-06-23:
That same PDF mentions "tutorial" style superdarn presentations.

ksterne, 2019-01-11:
@egthomas I thought I was remembering there being an issue with the script @sshepherd had where there were differences between license and licence or something else altogether that didn't have those words in them.  There hasn't been clear guidance from JHU/APL and the PI on this issue which I think might be holding it back.

egthomas, 2019-01-11:
There was one or two bugs that popped up but we fixed them immediately.  The license/license difference was missed by the rpkg scripts Rob originally used.  As I said before, the opportunity for discussion was unfortunately skipped during last year's PI meeting so now we have to wait until this year's workshop.

mts299, 2019-06-20:
Ran into Ethan Miller at CEDAR and asked him about this issue. He will be sending Kathryn and I an email to confirm that we can strip out the APL copyrights and replace it with SuperDARN. He did ask we keep the author names in the code. 

Will confirm further once Kathryn, and I receive the email and forward it to the co-chairs to give the go ahead.

mts299, 2019-08-06:
Ethan Miller got back to me and we cannot remove the APL copyrights which now being more versed in copyright laws and licenses that is fair because it is illegal. To remove copyright we need an agreement made of transfer of authorship which is taking credit away from the original author. Ethan suggested portion copyrighted, as in multiple copyright lines on shared creative works of code which is fair and Kathryn agrees with portions copyright. 

With this email and not fully understanding copyrights and licenses myself, I did some more digging on the matter by talking to a copyright coordinator at the University of Saskatchewan and talking to an ethics professor in the Computer Science department. 

These are my findings (no biases, just facts):

# Copyrights 
Copyright law is the exclusive legal right, given to an originator of the expression of an idea (protection on the expression, not the idea) to permit copying, modifying, distribution rights to others. 

**Note:** Licenses (a separate thing from copyright) can trump copyright permissions, this is how open-source code works. Will provide more information lower down. 

By law and probably a lot of peoples contracts, programmers are to include their/employers (which for most of us is probably our institution) copyright at the top of the file. At USASK we sign an agreement stating all intellectual property belongs to the university. So anyone at SuperDARN Canada has to copyright to the University of Saskatchewan. If a programmer does not copyright their work then it defaults anyways to the author of the code. So even if I do not include the copyright line in my code, by law the code is still copyrighted to me as the author. This is to protect anyone who is less informed of copyright laws. So even though we have not been doing it in RST, all the code everyone has written is already copyrighted to them. 

# Licenses 
Open-source licenses are based on one of the two concepts _copyleft_ or _permissive_ 
Both concepts allow the user to copy, modify and distribute the code without the copyright holder permission. However, _copyleft_ restrict the software to be included with other software of different licenses. This is to prevent offshoot derivatives of the work being used for proprietary or non-open-source purposes. So keeping the code open to everyone to use, which is the type of concept GPL uses. 
_permissive_ plays the thought of "Do whatever you want with my code" this is usually used in projects where developers just want their code out there and used and not worried how it is being used. _permissive_ licenses like MIT can be mixed in with other licensed code, allowing it to have more freedom. 

Besides the general concepts of licenses, every type of license have its own condition, permissions and limitations the developers and users of the software must uphold. If licenses are violated then a lawsuit can be made. 

If you want to look at the various open-software licenses and their permissions, conditions, and limitations this is a very good link: 
https://choosealicense.com/licenses/

Also, best practice for licensing code is to have the license file in the main directory of the repository. Next to the README.md is more common and GitHub usually suggests this when making a new repo. 

Side note: any public repositories on GitHub default to open-source license permission. By picking a license you are just ensuring what permissions are allowed and limitations assuming you also follow the conditions of the license. 

## GPL v 3.0 License

Permissions | Conditions | Limitations
-- | -- | --
Commercial use Distribution Modification Patent use Private use | Disclose source License and copyright notice Same license State changes | Liability Warranty

GPL v 3.0 also requires all code be open-source when distributing so you cannot closed-source it and only distribute executable code. Also, the authors of code cannot revoke their code from the software. 

https://www.gnu.org/licenses/licenses.html

# Summary
Copyrights with an open-source license are more for credit like putting your name on a paper and the institution you belong to. You do this to get the credit for the work you have done. The permissions are then passed and dictated by the license, similar to submitting to a journal, you are giving them the permission to copy and distribute your work so other researchers can cite it and aid in their research. 

# General questions: 

## Can APL revoke their code in RST?
No, as stated before the copyright with GPL v 3.0 license is more of credit to the author but all permission rights are given up once the author commits the code to a software package that has an open-source license that prevents revoking, which GPL v 3.0 does. Seeing how APL was also the one to pick the GPL v 3.0 license back in 2012 it would be really hard for them to violate their own license. However, if we violate the GPL v 3.0 license there is a chance they could revoke the code on the moral right (under copyright law) that we are not keeping the integrity of the code. 

## Can we have a SuperDARN entity for copyrighting? 
Yes and No, in theory, SuperDARN could become its own entity for copyrighting; However, this would require a lot of work to be done by the PI's to have it happen. Because we are a multi-country organization with no designated headquarters, the PI's would have to make an agreement that SuperDARN copyright includes the following institutions, that we are followin [... cut ...] issions, and limitations are similar or more restrictive of the previous license then this is doable without asking all authors of the code. If the new license is less restrictive like MIT (known as a weaker _permissive_ license) then we would need to get permission from all authors of the code including APL to change licenses. Because APL licensed their code under GPL v 3.0 we cannot license half of RST under another license because GPL v 3.0 requires all software to be under GPL v 3.0 license.

mts299, 2019-08-06:
Answering the main questions of this Issue:
1. license.txt or license.md should be GPL v 3.0 Since that is the license APL mentioned in their code in 2012. Which is a really good and commonly used license and requires all the code to be under the same license.   
https://github.com/SuperDARN/rst/blob/444abd4f38ff18bb6f9a25b6272254fe8af8b4d0/codebase/superdarn/src.bin/tk/plot/map_plot.1.12/plot_source.h#L8-L25 
2. No, we cannot remove the copyright lines of APL 

With that said maybe we could close this issue and make a PR to add a license.md file into our repo :+1:

ecbland, 2019-08-07:
A huge thank you to @mts299 for investigating this longstanding issue and providing a detailed explanation 💯. And thanks to the USASK copyright coordinator too. 

Unless @sshepherd or others have further questions/comments then I agree we can close this issue and make a new PR to add the license file.

mts299, 2019-08-20:
Closing as there has been no other discussion on this in the past two weeks :)

closed 2019-08-20 by mts299