SuperDARN/rst #34: RadarLoadHardware return value

ksterne, 2017-03-10:
Looking through some old notes, I saw https://github.com/SuperDARN/rst/blob/94d90a09d0af48db34d0f1623a9bddfbcc05b143/codebase/superdarn/src.idl/lib/legacy.1.6/rbposlib.pro#L102-L107  was changed.  I was under the understanding that RadarLoadHardware returned a zero if good and so the condition needed to be if it was not equal to zero.  However, in looking in the `codebase/superdarn/src.idl/lib/main.1.24/radar.pro` file, it seems as though the RadarLoadHardware routine will return zero if there's an IO error.  I'm not the best at IDL, so please correct me if I'm wrong.  Otherwise, should we cleanup how this is noted or more of a comment than "SGS for DLM" ?

sshepherd, 2017-03-10:
> rst/blob/94d90a09d0af48db34d0f1623a9bddfbcc05b143/codebase/superdarn/
> src.idl/lib/legacy.1.6/rbposlib.pro#L102-L107 was changed. I was under

This is an issue that needs to be sorted out later for the DLMs. The
problem is that the DLM version returns the opposite of what the C version
returns. I make the change myself when needed but haven't touched anything
in the DLMs, yet.

egthomas, 2017-03-30:
I think Simon may have meant the DLM and IDL versions return different values (rather than DLM and C).  Looking at the C code, RadarLoadHardware in radar.c of the radar.1.21 library will return -1 if an error occurs:

https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/radar.1.21/src/radar.c#L162-L163

This means that for a user with DLMs enabled, a call to RadarLoadHardware in IDL will return a -1 if an error occurs.  Otherwise if DLMs are not enabled, I'm not sure what RadarLoadHardware returns for an error since the iofail point at the end of the function isn't actually tied to an ON_IOERROR call.  Maybe 0 since a return value isn't set anywhere else?

egthomas, 2017-12-08:
Ok, so my previous post was not entirely accurate.  The IDL version of RadarLoadHardware will return a zero if everything works correctly:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.idl/lib/main.1.25/radar.pro#L248-L300

However the DLM version of RadarLoadHardware will return the number of radars loaded if it works correctly:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.dlm/rposdlm.1.9/rposdlm.c#L385-L403

@sshepherd's change to rbpos was appropriate for use with DLMs since the returned value would (probably) never equal zero.  But really, the return value of the DLM version of RadarLoadHardware should probably be changed to zero on success to make it consistent with the original IDL version.  Whether or not this is crucial for the 4.1 release is up to @ksterne.

egthomas, 2018-05-01:
Fixed in #125.

closed 2018-05-01 by egthomas