SuperDARN/rst #53: Hokkaido renaming to Hokkaido East

ksterne, 2017-04-24:
This is fairly minor, but I've gotten a request to rename Hokkaido to Hokkaido East to clear up some confusion.  So, this would be a change in:

https://github.com/SuperDARN/rst/blob/433f8d07da589ace41b176fa2aa82570083d2d62/tables/superdarn/radar.dat

Probably fairly minor and something we could change in the release branch next week if we wanted.  Otherwise, I can do a hotfix to the code now.

sshepherd, 2017-04-24:
> 70083d2d62/tables/superdarn/radar.dat

I suggest that any renaming of radar IDs first get passed along to the PIs.
This change will affect quite a bit of code and should not be done without
careful consideration. For instance, are we renaming all the existing
files? I would vote against such a change.

Simon

ksterne, 2017-04-24:
This is not changing the radar code letters.  Only "Hokkaido" -> "Hokkaido East". Does this require PI approval?

sshepherd, 2017-04-24:
Ah, just a change in radar.dat. No. This file likely has a few
inconsistencies in it. Evan fixed many, but I suspect more remain.

Simon

egthomas, 2017-04-24:
I still think we may need to run this by Nishitani-san.  Modifying the entry in the radar.dat file will likely alter the radar name appearing on everyone's output plots (using the RST code or otherwise).  Will the disappearance of the "Hokkaido" radar - which has been referenced in dozens of journal articles - lead to more confusion?

pasha-ponomarenko, 2017-04-24:
Well, I think there is more confucion now when one sees Hokkaido and Hokkaido West names in the same publication...

ksterne, 2017-04-25:
Hey @egthomas, this is in response to a request from Nishitani-san as you'll note the VT SuperDARN has changed over (aside from static plots).  Here is a snippet from his e-mail:

```Dear Kevin (and Paul),

.....

By they way, I have a concern about your way of identifying the radar
on your web server. The Hokkaido East is labelled as 'Hokkaido'. Before the
deployment of the Hokkaido West it was appropriate. However, the Hokkaido
West started operation around the end of October 2014. Then after that
the naming of 'Hokkaido' is rather confusing. Could you change it to the
'Hokkaido East'? Thanks in advance.

Best regards,

Nozomu Nishitani 
```

I'll make this change in the release branch after I create it tomorrow morning.

sshepherd, 2017-04-25:
Another question/potential issue with the radar.dat file that Kevin is editing here. The current file has lyr starting on 20161016, however there are numerous rawacfs (obtained from one of the "mirrors") that exist long before that date. The RST falls over when you try and use data from before 20161016. We either need to correct the start date or remove those files from the mirrors. Perhaps that has already happened over on the DDWG, if so, my apologies for bringing it up here.

egthomas, 2017-04-25:
Fair enough, just wanted​ to make sure the PI was in the loop! While you're at it, I think we should change the 'Svalbard' entry to 'SPEAR' to also avoid confusion with the new Longyearbyen radar.

ksterne, 2017-04-25:
@sshepherd, I'd contact the place that runs the mirrors to clear the file distribution issue.  Is that still the case?  If so, that's something to take up with the DDWG.  I think BAS may have had some early on, but I thought we'd gotten that straightened out.  Also I think there's a typo there that should be 20161019 as I've got an e-mail from Dag noting the official opening date on Oct. 19th and not the 16th.

@egthomas, is the SPEAR radar a SuperDARN radar?  I'm not sure if I've ever seen files from svb; could this be removed?  Otherwise, I can make the change as noted in the release branch.  I've got two other changes in the queue for the release branch as well.  I'll try to make it tomorrow afternoon EDT, but not promises that it might be earlier.

egthomas, 2017-04-25:
SPEAR was a Leicester radar/heater system on Svalbard. Not sure if it's still running, but I imagine it was added to the radar.dat file so the UK folks could manipulate the data using RST-ish software. Would hesitate to remove it entirely without input from someone like Mark Lester.

egthomas, 2017-05-01:
I've made the above changes for Hokkaido (East) and SPEAR in the radar.dat file on the release-4.0 branch (dceebbc672dc1372a46d5ea972cb76bc53fa271c).  Closing this issue.

closed 2017-05-01 by egthomas