SuperDARN/rst #56: Builds for fitacf.2.x and fitacf.3.0

kkotyk, 2017-05-01:
I'm going to need a strategy to build both these libraries. RST will only build fitacf.3.0 if the directory is called that, based on the convention. I'm thinking about calling the directory fitacf.v.3.0 or something. Any real preference here?

egthomas, 2017-05-01:
If FITACF 3.0 is the version which will future receive updates then maybe it should follow the standard convention (ie `src.lib/tk/fitacf.3.0` and the older 2.x library should be changed to something else (like the `src.lib/tk/fitacfT.2.4` we already have).

pasha-ponomarenko, 2017-05-01:
I am not sure if I understand what the problem is... Is it related to the compilation process?

kkotyk, 2017-05-02:
Yes. Typically RST will only build the latest version of a package. In this case we want to build both fitacf.2.4 and fitacf.3.0. We will need to use a trick to get both to compile.

sshepherd, 2017-05-02:
FYI - this is the reason for naming aacgm_v2, mlt_v2, igrf_v2, etc. so we
could have both versions available at the same time.

kkotyk, 2017-05-02:
Okay well we could go with fitacf_v3, or we could go with Evan's idea. I'm not opposed to either.

egthomas, 2017-05-02:
I think the v2 naming scheme made sense in the context of AACGM but am not sure it should be applied to FITACF.  It's your call though - we're just anxious to start using it!

kkotyk, 2017-05-04:
I've been looking a bit closer after I get both 2.5 and 3.0 to compile, I think I can get make_fit to use both versions. I can add a flag like -version which you can specify 2.5 or 3.0 with one of them being default. Two questions: do you guys think I need to make a new version of make_fit(1.19), or are we going to use git revisioning to track changes to RST components? The second question might be for @pasha-ponomarenko, but what version of fitacf did we agree to use as default?

egthomas, 2017-05-16:
Are you still looking for feedback here?  I'm not sure that we need to keep incrementing the version numbers of all of the libraries/routines, except for specific cases like the FITACF library which people often cite.  I think we also need to be careful about keeping track of the major/minor revision numbers being written to the files when we do update those version numbers.  Although maybe those are meant to be unrelated?

kkotyk, 2017-05-16:
Okay makes sense. I think I checked awhile back, but you incremented version numbers of gridding stuff? I've made sure to update the revision major/minor number for the fitacf stuff.

egthomas, 2017-05-16:
We just had a quick chat about this and were thinking it might make sense to still increment version numbers of functions or libraries if there has been some change in the output or functionality since the previous release.  So for example, if there were multiple minor changes in the rpos library we would update the version number once from 1.7 to 1.8 for the next release.  That way the versioning is still apparent for non-git users or for us in the future if we were to stop using git.

I did make the jump from 1.xx to 2.0 for make_grid and the gtable library on the `feature/grid_v2` because I felt that major upgrades had occurred since the previous version.

kkotyk, 2017-05-16:
Okay sounds good.

egthomas, 2017-07-05:
@kkotyk - since the branch with fitacf3 appears to be fully integrated (data quality issues notwithstanding) can we consider this issue (and #55) closed?

egthomas, 2017-10-05:
Closing this since the original issue of building both fitacf.2.5 and fitacf.3.0 libraries appears to be solved.  @kkotyk can re-open if he feels otherwise.

closed 2017-10-05 by egthomas

kkotyk, 2017-10-05:
Hey apparently I missed the email about this in July. Yes this seems fine to me.