SuperDARN/rst #69: doi thoughts?

egthomas, 2017-06-02:
Given the new wave of citability / reproducibility of software being recommended for journal submissions I was able to use Zenodo to attach a doi to the new RST 4.0 release:

[image]

It would have been nice to include this in the readme of the tagged 4.0 release, but unfortunately it seems like a chicken-and-egg scenario where you can't get the doi until after a release is published.  I could still go ahead and add this to the readme on the master branch (as a hotfix?).

What do people think about this?  Good idea?  Bad idea?  Indifferent?  Pipe down and let us finish our workshop presentations?

pasha-ponomarenko, 2017-06-02:
I would leave it until the WG meeting on Tuesday night.

egthomas, 2017-06-02:
Fine by me.  One thing I should point out is that we can easily modify the author list (ie so it includes more than just my name).

pasha-ponomarenko, 2017-06-22:
Mark is looking into it. There are some issues to resolve before we can proceed.

egthomas, 2017-08-04:
As a temporary solution (since I was uncomfortable with being the only name listed) I've modified the citation on the Zenodo doi to read the following:

SuperDARN Data Analysis Working Group. Participating members, Thomas, E. G., Ponomarenko, P. V., Bland, E. C., Burrell, A. G., Kotyk, K., Shepherd, S. G., and Sterne, K. T. (2017), SuperDARN Radar Software Toolkit (RST) 4.0, doi:10.5281/zenodo.801459.

I tried to follow the authorship style for models / software packages like IGRF, which use something like this:

International Association of Geomagnetism and Aeronomy, Working Group V-MOD. Participating members, Finlay, C. C., Maus, S., Beggan, C. D., Bondar,...blah blah blah so many coauthors..., Wei, Z. and Zvereva, T. I. (2010), International Geomagnetic Reference Field: the eleventh generation. Geophysical Journal International, 183: 1216–1230. doi:10.1111/j.1365-246X.2010.04804.x

Let me know if anyone has any issues - this could also still be overruled by a decision from Mark.  Again, the link is available here:

[image]

egthomas, 2017-08-08:
I'm closing this issue for now - we can revisit it if/when any decisions are handed down from above.

closed 2017-08-08 by egthomas