SuperDARN/rst #73: Sessai's concerns about removing ROS routines

pasha-ponomarenko, 2017-06-15:
At the SD'17 meeting Sessai expressed his concern about the removal of ROS-related code from RST becasue it makes much harder to synchronise modifications to FITACF and other related rountines, which are used at the site for a quick data quality check or real-time monitoring. 
Any thoughts how to resolve this issue? 
It would also be a good idea to add Serssai to the discussion list.

kkotyk, 2017-06-15:
Well I don't know about other institutions, but we are still running fitacf.2.0 on our radar, so it's not like updates and synchronizations are currently being done. We would need to develop a new technique for delivering analysis software to the radars, and to somehow make sure everyone is using it. Is everyone running the same version of programs like rawacfwrite? What strategy was used to make sure everyone was running the latest code back in the older days when there were more modifications?

egthomas, 2017-06-15:
Does anyone know if there has been any action in the Operating Software Working Group?  Maybe we need to start coordinating with them on shared routines like fitacf.

@kkotyk - I think there's been a lot more fragmentation of the operating software than the analysis software over the years with various styles of radar (e.g. Stereo, MSI, digital, etc).   You can see the results of Mike's OS audit from a few years ago here:

http://vt.superdarn.org/tiki-download_file.php?fileId=1333

egthomas, 2017-06-15:
Interestingly, the charter of the OS-WG found on the VT webpage states the following:

> Responsibilities:
> 1) Set the contents of the hdw.dat files for the radars and distribute updates.
> 2) Assign ids for new radars, disseminate information about changes to the hdw.dat file.

pasha-ponomarenko, 2017-06-16:
Well spotted, Evan! We should get in touch with Mike, but I have a gut feeling that all this hdw.dat business will be softly pushed back to us anyways...
;-)

egthomas, 2017-10-05:
So is this still an issue?  It sounds like most people are running outdated versions of fitacf on their radars anyways, and I'm not sure the OS code which we stripped out before the 4.0 release was compatible on all of the radars either.  If the OS-WG ever gets their act together then we could start coordinating with them.

egthomas, 2018-03-22:
Closing this as it's really an OS-WG task as discussed during the telecon.

closed 2018-03-22 by egthomas