SuperDARN/rst #77: macOS installation

egthomas, 2017-06-19:
I think we're all aware that one of our biggest outstanding issues is the installation of the RST on systems running macOS.  Prior to the 4.0 release, @aburrell was eventually able to install the RST after a lot of headache and modifications to the build/environment scripts and I was unable to get `make.code superdarn rst` to progress beyond the fontdb.1.9 library.

Bill Bristow sent an email on Friday saying he was able to install the RST on his Mac desktop but had to fix numerous compiler warnings - specifically the ones related to printf/fprintf/sprintf statements (which have all been fixed in #75).  He also said that he needed to add a symlink to the X11 include directory in /usr/include: `ln –s /usr/X11/include/X11 X11`

So for any Mac users out there who want to try compiling the RST, I would suggest trying the `develop` branch which now has the printf warnings taken care of.  You may also want to try Bill's suggestion about adding the X11 symlink.

aburrell, 2017-06-19:
This is a good idea.  I needed this symbolic link for something previous, so already had it and completely forgot it existed.  Once we get it going we should try and track down a fairly new mac and do an install.

egthomas, 2017-07-05:
So has anyone else given this a try since the workshop?

agrocott, 2017-07-06:
I have just successfully* installed to a 2013 i7 Mac Mini running Sierra 10.12.5.  I had a couple of issues with header files not being found (cdf.h and a couple of png/libpng.h files) so I found where they were hiding and copied them into the directories that the compiler seemed to be looking for them - not very elegant, perhaps, but it seems to have done the job.

I also couldn't create the symlink to X11 in "/usr/include" due to Apple's "System Integrity Protection". Apparently you can turn it off, but instead I created the link in "/opt/local/include" which is not protected and seems to be in the compiler path.

*so far things seem to work but I've not done any extensive testing...

egthomas, 2017-07-06:
Thanks Adrian!  Do you know if you used the `develop` branch (with Bill's fixes) or the master / 4.0 version we sent around over the darn-users list?

agrocott, 2017-07-06:
I used the develop branch - I got the same print related compiler warnings with the master version.

aburrell, 2017-07-11:
Right.  I just finished testing this on another mac and doing a web search.  Best I can conclude is that the problem lies with the implementation of bash on iTerm.app.  I tried several things to fix this, without success.  What did work was doing the installation on XQuartz.  I just tested this on the master branch and once I fixed the fprintf statements that implicitly assume a formatting statement there were no issues with installation.

egthomas, 2017-07-11:
Did you still have problems with the IDL skip in make.code or libraries being compiled out of order?

aburrell, 2017-07-11:
No.  None of the problems popped up.  All of the IDL instructions were followed without me having to do anything.

It’d be nice if someone could test this on their machine and also if someone could read over the updated README.md instructions and see if they need clarification.

ksterne, 2017-09-13:
Saw this was closed without much note here.  I know @sshepherd is a mac user, were you able to get things installed easily?  @agrocott, @aburrell, @egthomas is this an issue anymore?  Just checking here.

agrocott, 2017-09-14:
No further issues noted here, thanks.

egthomas, 2018-08-08:
@agrocott has raised the issue that he needs to manually set the `DYLD_LIBRARY_PATH` environment variable on his Mac for the code using the CDF library (eg `map_addimf`) to work.  Can any Mac users verify whether a change to `rst.bash` similar to the lines below would work?

```
if [ "$OSTYPE" == "darwin" ]
then
    export DYLD_LIBRARY_PATH="${LIBPATH}:/usr/local/cdf/lib" (or something else?)
else
    export LD_LIBRARY_PATH="${LIBPATH}:/usr/local/cdf/lib"
fi
```

@arfogg - was this the CDF problem you had encountered?

egthomas, 2018-08-08:
@arfogg - sure!

mtwalach, 2018-08-21:
Hi @egthomas!
I have `export LD_LIBRARY_PATH="${LIBPATH}:/usr/local/cdf/lib"` in my `rst.bash` file but no dynamic library path. I think this might be something to do with the newer OS which I haven't upgraded to.. I do remember making a number of symbolic links so it would find all the CDF libraries, so it could also be something to do with this?
Sorry if this is not overly helpful...

asreimer, 2018-08-22:
Just wanted to note here that I was able to get RST to install on macOS using only Anaconda. By this I mean I didn't use macports nor did I use brew.

Planning on posting full details later this week. Will link back here after I post.

asreimer, 2019-01-22:
Still needs to be added. Thanks for the ping.

mtwalach, 2019-01-28:
Hi!
I am running macOS Sierra at the moment and I used homebrew to install both the libhdf5 and cdf packages. I also had to make some symbolic links to the cdf and X11 libraries because it couldn't find them by itself even though I set up the paths to the folders. 
Let me know if there are any particular details you are interested in or if you need to know more!

aburrell, 2019-01-28:
I also use macports.  I believe macports has all the HDF5 and CDF dependencies, but I could be remembering wrong...

egthomas, 2019-04-10:
Please continue macOS installation discussions in #215.

closed 2019-04-10 by egthomas