SuperDARN/rst #81: Insight on alternate lag 0

kkotyk, 2017-06-29:
I just was just discussing with Pasha and we were curious on if anyone has any insight to which places the alternate lag 0 in a .fitacf file lag table might be used.

sshepherd, 2017-06-29:
Isn't it used to compute lag0 power at ranges beyond the critical range,
i.e., the range at which the first pulse interferes with the second?

pasha-ponomarenko, 2017-06-29:
Then it should be done inside ROS, I guess.

kkotyk, 2017-06-29:
Okay I've notices this because FITACF seems to generate an alternate lag 0 for some files, when it doesn't always exist in the rawfile. This seems to be the case for 2.5 or 3.0.

kkotyk, 2017-06-29:
I see the problem. FITACF always assumes the lag table in the rawacf file has an alternate lag 0, which doesn't appear to be the case. This seems to lead to some odd values for an alternate lag 0 in the fitacf file in this case.

sshepherd, 2017-06-29:
My recollection is that the alternate lag0 is specified in the control
program. Here is an example from my normalscan control program:

  int ptab[8] = {0,14,22,24,27,31,42,43};
  int lags[LAG_SIZE][2] = {
    { 0, 0},    /*  0 */
    {42,43},    /*  1 */
    {22,24},    /*  2 */
    {24,27},    /*  3 */
    {27,31},    /*  4 */
    {22,27},    /*  5 */

    {24,31},    /*  7 */
    {14,22},    /*  8 */
...
   {43,43}};   /* alternate lag-0  */

Simon

kkotyk, 2017-06-29:
Okay makes sense. So here's a trouble file for me. I'm looking at 20070101.2201.00.sas.rawacf. This file was apparently generated from fixrawacf which I'm wondering if it caused errors. It's comf field says Id: normalsound.c,v 2.4 2006/04/13 18:14:36 barnes Exp $ which shows the radar was running v2.4 of normalsound. The comments claim in version 2.3 you added a modification to the lag table @sshepherd. Do you remember what that was Simon? The control program shows a alternate lag 0. I'm wondering if the fixrawacf program dropped it.

sshepherd, 2017-06-29:
Me?! I can't remember what I did last week let alone a decade ago, but I
certainly don't remember editing anything that would run on the Canadian
radars. It must have been when I was still at APL, which was before 2000.
My money is on the fixrawacf program.

kkotyk, 2017-06-29:
Fair enough. I thought it was worth a shot to ask. This isn't the first time I've found issues with fixrawacf produced files. I would suggest that these files be identified and maybe looked at. Sometimes they are entirely missing fields or have empty arrays.

sshepherd, 2017-06-29:
I am not familiar with fixrawacf. Is this a utility from the original RST
or something that we've recently introduced?

kkotyk, 2017-06-29:
I don't think it was an RST utility, but I think it was a program someone wrote years ago as a separate utility to edit rawacfs to fix issues. I just asked Kevin K. about these files and apparently this is an open issue for the DDWG.

kkotyk, 2017-06-29:
We probably don't have to dive into this issue too deeply here as it was not the intended purpose of this question.

egthomas, 2017-10-06:
Are you still looking for insight on this issue @kkotyk?  At this point you may not be able to get a definitive answer until the next workshop.

kkotyk, 2017-10-06:
Not specifically on this, but this issue led me to find that the raw data doesn't always data at every range gate. Pasha and I were wondering what criteria was used for filtering data on the radar before the fitting stage. I could close this.

closed 2017-10-06 by kkotyk