SuperDARN/rst #82: Loading custom colour tables

ecbland, 2017-07-03:
The way in which colour tables are loaded in e.g. `time_plot.c` and `field_plot.c` using `-pkey`, `-vkey` etc. has changed. Previously it was possible to load colour tables from any directory (incl. outside RST) by specifying the full path to the file, but now the code only looks for colour tables in the default directory, `$RSTPATH/tables/base/key/`. 

For example, line 643 in [time_plot.c](rst/codebase/superdarn/src.bin/tk/plot/time_plot.1.8/time_plot.c) reads:
`if (key_path == NULL) key_path = getenv("COLOR_TABLE_PATH");`

Since there is no option to set `key_path`, RST cannot load colour tables unless they are placed in the default COLOR_TABLE_PATH directory. Is this the intended functionality?

egthomas, 2017-07-05:
I can't speak for @sshepherd since he was the one who made those changes, but my guess is the original intention was to remove the need to type out the full path in front of the color table filename each time one calls a plotting routine.  It sounds like `key_path` should also be added as a command line option (maybe with its default value set to `COLOR_TABLE_PATH`).

As a temporary workaround can you copy your external color tables into the `COLOR_TABLE_PATH` (ie `rst/tables/base/key`?

ecbland, 2017-07-05:
I agree that this bug can be fixed by adding `key_path` as a command line option - perhaps this was simply an oversight when the code was modified.

Yes, my temporary work-around was to copy my custom colour tables into `$RSTPATH/tables/base/key`.

sshepherd, 2017-07-05:
I'll try and have a look at this today (yesterday was a holiday of sorts).
Am I correct in thinking that you would prefer that we revert to how things
worked before I made the changes? I recall the changes were a result of not
wanting to specify a full path for a color table in the command line
options.

Simon

egthomas, 2017-07-05:
Couldn't we have separate options like `key_path` and `key_name`?  That way users who want to use the default RST color tables have less to type but still have the flexibility to use an external color table too?

ecbland, 2017-07-05:
I agree with Evan :)  Currently the "key_name" is given by the command line options `-vkey`, `-pkey` etc, so we'd just need to add `-key_path` and update the documentation for the affected routines.

sshepherd, 2017-07-07:
Hi Emma, Evan and others,

Sorry for the delay. I remember when I made this change. It seemed rather
unsavory to have to specify the full path of the color table, even the
default, so the environment variable seemed like a good option. I didn't
anticipate anyone using color tables other than those provided in the RST,
but I guess it doesn't surprise me based on the variety of color schemes on
display in Italy.

The solution of allowing for a command line option for the path seems
reasonable, however, some of the plotting routines (time_plot, for
instance) has several color keys:

-pkey pkeyname       load the power colorkey from the file pkeyname.
-vkey vkeyname       load the velocity colorkey from the file vkeyname.
-wkey wkeyname       load the spectral width colorkey from the file
wkeyname.
-fkey vkeyname       load the frequency colorkey from the file fkeyname.
-nkey vkeyname       load the noise colorkey from the file nkeyname.

Should each one of these have a corresponding key_path or just one?

I am not git-savy enough to be able to look at how the code was written
before I got my hands on it and Evan is out of the office today, so he
can't help me.

I think the suggestion is the modify all of the plotting routines to allow
for a key_path filename to be passed in through the command line and the
code modified to something like:

...

OptionAdd(&opt,"keypath",'t',&key_path);

...

if (key_path == NULL)                        /* key_path has not been set
on the command-line */
   key_path = getenv("COLOR_TABLE_PATH");    /* so use the environment
variable.              */

...

Is that what is being suggested?

Simon

closed 2017-09-06 by sshepherd