SuperDARN/rst #83: fitacf missing error output

ksterne, 2017-07-10:
From something that was raised in the DDWG, it appears as though that make_fit processes the 20110204.22*pgr.rawacf file as it does not throw any kind of error message.  However, the fitacf file ends shortly after 23:17 utc.  There should be some kind of error message thrown here if it's crashing.  make_fitex2 does throw a nasty error referring to munmap_chuck() when trying to process this file.

egthomas, 2017-07-11:
I get a segmentation fault when using the `master` branch version of make_fit on that file.  Here is the verbose output:

```
ferguson:~$ make_fit -new -vb 20110204.2201.00.pgr.rawacf > 20110204.2201.00.pgr.fitacf
2011-2-4 22:1:0 beam=15
...
2011-2-4 23:17:0 beam=15
2011-2-4 23:17:3 beam=14
2011-2-4 23:17:6 beam=13
2011-2-4 23:17:9 beam=12
2011-2-4 23:17:12 beam=11
2011-2-4 23:17:15 beam=10
2011-2-4 23:17:18 beam=9
2011-2-4 23:17:21 beam=8
2011-2-4 23:17:24 beam=7
2011-2-4 23:17:27 beam=6
2011-2-4 23:17:30 beam=5
2011-2-4 23:17:33 beam=4
2011-2-4 23:17:36 beam=3
2011-2-4 23:17:39 beam=2
2011-2-4 23:17:42 beam=1
Segmentation fault
```

make_fitex2 (also on the `master` branch) gives the following result:
```
ferguson:~$ make_fitex2 -new 20110204.2201.00.pgr.rawacf > 20110204.2201.00.pgr.fitex2
2011-2-4 22:1:0 beam=15
...
2011-2-4 23:17:0 beam=15
2011-2-4 23:17:3 beam=14
2011-2-4 23:17:6 beam=13
2011-2-4 23:17:9 beam=12
2011-2-4 23:17:12 beam=11
2011-2-4 23:17:15 beam=10
2011-2-4 23:17:18 beam=9
2011-2-4 23:17:21 beam=8
2011-2-4 23:17:24 beam=7
2011-2-4 23:17:27 beam=6
2011-2-4 23:17:30 beam=5
2011-2-4 23:17:33 beam=4
2011-2-4 23:17:36 beam=3
2011-2-4 23:17:39 beam=2
2011-2-4 23:17:42 beam=1
*** Error in `make_fitex2': double free or corruption (out): 0x00000000024196f0 ***
======= Backtrace: =========
...
gobbledy gook
...
```

kkotyk, 2017-07-11:
I tried running this and it doesn't crash or throw any sort of error for me. My guess is that there is a low level bug with the file reading routines?

egthomas, 2017-07-18:
Just a follow up, but I get segmentation faults right after the `23:17:42 beam=1` record using versions 2.5 and 3.0 of FITACF with make_fit on the `add_fitacf.3.0` branch.  According to dmapdump that is also the final record in the rawacf file.  Are we sure we're all working from the same rawacf file?

aburrell, 2017-07-18:
I’ll create a link to the file I’m working with tomorrow

ksterne, 2017-08-04:
As noted in #90, I'm betting my note about not seeing a seg fault is related to the code I was using while processing this file.  I'll try to redo things and see if I get a similar error message as @egthomas posted.

egthomas, 2018-03-13:
This file (20110204.2201.00.pgr.rawacf.bz2) was blacklisted as noted in DDWG issue # 24 on 20 July 2017.

closed 2018-03-13 by egthomas