SuperDARN/rst #85: make_grid, scan flag, and "-is" option

egthomas, 2017-07-13:
Ok, so under certain scenarios (usually Special or Discretionary Time experiments) make_grid will throw away all of the fit records in a "scan" interval according to some criteria related to the scan flag.  Not only will this throw away potentially usable data, but it will also cause NANs to be written to the frequency, noise_mean, and noise_sd parameters in the grid file due to divison by zero in GridTableMap.  The guilty function is exclude_outofscan in make_grid.c; this function can be disabled by setting the command line `-is` option.  I think the intention is to throw away data with scan flags set to -1, however there is no documentation in the source code or in the doc.xml file about why this was ever set as the default behavior.

Does anyone have any experience with using this flag?  Or even why the scan flag is sometimes set to -1?  I need to do some more testing, but my first thought was to reverse the behavior of the `-is` option such that you need to explicitly set it to throw away data with negative scan flags.

An example case can be found starting around 12:00 UT on 20100304 on Hankasalmi channels A and B when what looks like a heater experiment was running.  (some of this should be thrown out due to the frequency hopping but that's another known but unrelated issue)

kkotyk, 2017-07-13:
We had problems with scan flag = -1 back in 2015 using ZHO data. We never found out why some radars use an unconventional scan flag. I know that the davitpy group also had problems processing this data too. After talking with @ksterne, we added the -tl arg to make_grid in order to ignore scan flag.

egthomas, 2017-07-13:
Unfortunately setting the `-tl` flag isn't enough to sidestep this issue - if every beam sounding in a fit file is marked with scan flag = -1 then by default all of the data will be thrown out by the exclude_outofscan function in make_grid (https://github.com/SuperDARN/rst/blob/feature/grid_v2/codebase/superdarn/src.bin/tk/tool/make_grid.2.0/make_grid.c#L109-L116).

kkotyk, 2017-07-13:
That could make sense. At the time Kathryn was struggling to produce map plots that matched those on VTs website. We contacted @ksterne to use the same options they use to make maps on their servers backend.

egthomas, 2017-07-13:
Upon further inspection, the only radar this seems to affect (at least in the Northern Hemisphere) is Hankasalmi during stereoscan (6401/26401) operation on a variety of dates from 2010-present.  So there must be something unique to that control program causing the scan flag to be set to -1 a lot of the time.  The grid files on the VT server also show NANs for the frequency, noise_mean, and noise_sd parameters on those dates.

kkotyk, 2017-07-13:
The file we had problems with back in the day was 20140531.C0.zho.fitacf if you need another to test.

egthomas, 2017-07-14:
Yeah this looks like the same problem you guys encountered before (https://github.com/vtsuperdarn/VTRST3.5/issues/11) - stereo modes appear to set the scan flag to -1 rather than +1.  This can get particularly nasty on a camping beam channel where the scan flag on every sounding is -1.

This is partly why at the workshop I recommended creating a "core" set of grid files using an agreed upon set of make_grid options - it can be difficult for users to know why the code fails due to these odd cases like negative scan flags.  Maybe we want to make something like `-tl 120 -i 120 -is` the default behavior (while still leaving flexibility for users to change those options if they want).

egthomas, 2017-07-25:
I've updated the behavior of `make_grid` to fix these issues on the `feature/grid_v2` branch so I'm closing this issue.

closed 2017-07-25 by egthomas

ecbland, 2017-08-11:
I can partially answer the question about the negative scan flags, as we recently discovered the same issue on the LYR radar! 

In our standard radar control program, the negative scan flag is used to indicate that the radar **scans backwards (i.e. from beam 15 to 0)**. Our radar control program was adapted from a HAN control program, so the behaviour was simply copied across to LYR.

Presumably this is non-standard use of the scan flag, which has been implemented on a small number of radars (inadvertently in our case). Someone from Leicester might know why it was used this way. 

For the record, on 31 July we modified the LYR radar control program so that new scans are flagged with +1.

egthomas, 2017-08-11:
Thanks @ecbland!  That makes a lot of sense, and also sounds vaguely familiar about negative scan flags indicating a reverse behavior.