SuperDARN/rst #88: Changes in IDL paths is not reflected in idl.bash

pasha-ponomarenko, 2017-07-25:


pasha-ponomarenko, 2017-07-25:
"... are not reflected..."

egthomas, 2017-07-25:
Could you be a little more specific?  Not sure I understand the issue here.

pasha-ponomarenko, 2017-07-25:
Look at /rst/.profile/idl.bash on the develop branch. All paths are incorrect:

export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_STARTUP="${RSTPATH}/idl/startup.pro"

egthomas, 2017-07-25:
Sorry, I'm still confused.  Which of these is incorrect and what should they be changed to?  At a minimum, the `IDL_PATH` environment variable is going to be unique to each users' IDL installation directory.

kkotyk, 2017-07-25:
As far as I see there is no longer a idl folder in the top level, so the IDL_STARTUP path doesn't exist. We should maybe add a line to the installation instructions if the IDL_PATH needs to be manually updated.

pasha-ponomarenko, 2017-07-25:
These paths reflect the old directory structure like that on master-old branch. For exmple, ther is no idl folder in the root directory of the newer branches. The same relates to dlm.

pasha-ponomarenko, 2017-07-25:
I am talking about the internal directory structure of RST.

pasha-ponomarenko, 2017-07-25:
Oh, Keith already worte this!

pasha-ponomarenko, 2017-07-25:
I think this is how it should be corrected now:

#export DLMPATH="${RSTPATH}/dlm"
export DLMPATH="<IDL_DEFAULT>:${RSTPATH}/codebase/superdarn/src.dlm"

# IDL environment

#export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/codebase/superdarn/src.idl/lib"
#export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/codebase/superdarn/src.dlm"
#export IDL_STARTUP="${RSTPATH}/idl/startup.pro"
export IDL_STARTUP="${RSTPATH}/codebase/superdarn/src.idl/startup/startup.pro"

egthomas, 2017-07-25:
The top-level idl folder is created during compilation if your `IDL_IPATH` environment variable is set correctly and the idl_export.h file necessary for DLMs can be found.  See

https://github.com/SuperDARN/rst/blob/master/build/script/make.code#L164-L182

That was part of our fix for automatic compilation for systems without access to IDL.  If you do have IDL, step 1 of the readme instructions explicitly tells users to verify their `IDL_IPATH` environment variable.

kkotyk, 2017-07-25:
Okay so Pasha you just need to change that line in your file and it should be fine then. Try test that out.

pasha-ponomarenko, 2017-07-25:
Thanks Evan, it works now! 
Sorry for the hussle!

closed 2017-07-25 by pasha-ponomarenko

sshepherd, 2017-07-25:
Let us all know how you are getting on with this Pasha.

pasha-ponomarenko, 2017-07-25:
All in good time, Simon!
;-)

sshepherd, 2017-07-25:
I was just getting worried, Pasha. We hadn't had an update in something
like 30 seconds. ;>