SuperDARN/rst #91: Median filtering in make_grid routines

ecbland, 2017-08-11:
I have a question about the routine `FilterRadarScan` (rst/codebase/superdarn/src.lib/tk/filter.1.8/src/filter.c). 

When median filtering is done on the power, spectral width or lag0 power parameters, the `median` array gets sorted according to the **velocity**.  For example, on lines 632-638 we have:

`/* Sort velocity values in median Radarcell structure
                 * from most negative to most positive */
                qsort(median,cnt,sizeof(struct RadarCell *), FilterCmpVel);

                /* Set current beam/gate width to the center of the above
                 * array sorted by velocity (ie the median) */
                dst->bm[bm].rng[rng].w_l=median[cnt/2]->w_l;`

Surely the intention here is to sort by the spectral width using the subroutine `FilterCmpWdt`, or else we don't actually get the median value for the spectral width. `FilterCmpWdt` and `FilterCmpPwr` are defined but not used (lines 57 and 48 respectively), so perhaps this is a copy-paste error? There is no equivalent routine for the lag0 power.

egthomas, 2017-08-11:
You're absolutely correct that this does not behave intuitively.  I think the intention was to try to get the power / spectral width / lag0 power associated with the median velocity value that was selected earlier.  However, since values are discarded according to a 2-sigma criteria for each of those parameters, the same set of line-of-sight measurements might not be available for sorting, thus you might end up with a power or spectral width associated with a different velocity measurement.  

Changing the code to sort by power, spectral width, or lag0 power for those stages of FilterRadarScan would probably make more sense even if it changes the original intent.  It is worth noting that those extra parameters (power and spectral width) are only stored in the grid file if you set the `-xtd` option (and the lag0 power isn't stored at all).  Which raises the question - do people use the power or spectral width values stored in grid files?

egthomas, 2017-08-11:
It's also entirely possible that it was just a copy-paste error and I'm wildly speculating about the original intent behind always sorting by velocity to select the median value.

pasha-ponomarenko, 2017-08-11:
I guess the intention here might be to use the spectral width value which corresponds to the median velocity.

pasha-ponomarenko, 2017-08-11:
This is actually confirmed in the respective comment:

> /* Set current beam/gate **width** to the center of the above array sorted by velocity (ie the median) */

Note the use of "width" which effectively excludes the possibility of a cut-and-paste error.

egthomas, 2017-08-11:
@pasha-ponomarenko - I wrote those comments based on my interpretation of the code; unfortunately they don't tell us anything about what Mike or Rob intended.

pasha-ponomarenko, 2017-08-11:
Oh, I see!

pasha-ponomarenko, 2017-08-11:
I recall from either conversation with Mike or from reading their 1995 paper that median was applied in order to use the actually measured velocity vaue rather than some kind of its average. So, within this paradigm the use of the respective spectral width seems to make more sense...

egthomas, 2017-08-11:
That's possible.  I should also admit that my earlier response was narrowly focused on filtering and its relationship to `make_grid`.  I believe you can actually run the same filtering code using `fitfilter` on a fit/fitacf file to generate median filtered results without doing all of the other gridding steps.

ecbland, 2017-08-14:
> Which raises the question - do people use the power or spectral width values stored in grid files?
A current project at UNIS is using the spectral width parameter from the extended-format grid files. 

As @egthomas pointed out, the current code does not necessarily give the power/width corresponding to the median velocity due to the 2-sigma criterion. Therefore, the code does not actually do what the author(s) intended. Perhaps we should find out/agree on the desired behaviour of this routine and update it?

pasha-ponomarenko, 2017-08-14:
Well, I guess we need to confirm the original concept and goals of the "weighted median" filtering with Mike and then proceed with corrections.

I also wonder haw the gridded SW is used. The point is that, in contrast to the line-of-sight velocity,  it is rather sensitive to even slight variations in the fitting procedures so that its magnitude for the same data can vary significantly between different versions of FITACF. Moreover, overestimation of spectral width was the main reason for producing FITACF2.

egthomas, 2017-08-15:
The user can choose which parameters to median filter (velocity, power, spectral width, and/or lag0 power) by setting input `int mode` value accordingly.  So I guess the question is this - should `FilterRadarScan` operate on each parameter independently, or always return the parameter(s) associated with the range-beam cell providing the median filtered velocity?  Because right now it tries to do both.

egthomas, 2017-08-29:
I didn't have a chance to ask Mike about this when we were in Montreal but I can try to pin him down when I visit VT in about 2 weeks.

egthomas, 2017-09-14:
Mike and I just spent a while looking at this and he explained that the original intention was to provide a sensible power / spectral width value associated with the distribution of velocity values being considered while gridding.  He agreed there should also be an option to median filter each parameter independently rather than relating them back to the velocity distribution within the 3x3x3 window.  

It should be fairly straight forward to make that change - any suggestions for naming the command line argument which controls this functionality?

pasha-ponomarenko, 2017-09-14:
Please educate me how the filtered power and spectral width values are used further down the track? Weighting or something else?

egthomas, 2017-09-14:
The filtered power and spectral width values are not used again in the gridding procedure after they're calculated, other than being written to the grid file if the `-xtd` option for extended output is set via the command line.

egthomas, 2017-09-14:
@ecbland - I've added the capability to sort each parameter independently of the velocity distribution to FilterRadarScan in commit 70c9bca.  Let me know if you see any other changes that need to be made.

pasha-ponomarenko, 2017-09-14:
Thanks, @egthomas! So, it is just for illustrative purposes or for some sort of crude quality control...

egthomas, 2017-10-05:
@ecbland - did the above fix address your issue here?

ecbland, 2017-10-05:
@egthomas: sorry for the delay - I haven't had a proper look at it yet. I'll do it first thing tomorrow.

egthomas, 2017-10-05:
@ecbland no worries, I was just looking through some of the issues on here and hoping we could put a few of them to rest.

ecbland, 2017-10-06:
@egthomas: yes, your changes address the issue and work as described. Very helpful, thanks.

egthomas, 2017-10-06:
@ecbland great!  Can we consider this issue closed then?