SuperDARN/rst #95: 4.1 release goals and timeline

egthomas, 2017-08-29:
Since we're coming up on 3 months post-4.0 I thought it might be a good idea if we start laying out our goals for the next release.  At the workshop we had discussed having version 4.1 ready around this time (September) but I think December is probably more realistic given our previous pace.  Or would we rather wait until next year's workshop for a new release?

Here are the key pieces I think we should have ready for the next release:
1) revised version of FITACF library
2) revised version of gridding software
3) revised version of Map Potential software
4) better Mac support

There are currently outstanding pull requests for items 1 and 2 on the `add_fitacf.3.0` and `feature/grid_v2` branches respectively, both of which I personally feel could use some additional testing from additional users.  @sshepherd  and @aburrell have gotten started on items 3 and 4 on the `feature/map_potential_v2` and `macOS_installation` branches respectively.  I know there is still a lot of work we need to do on the Map Potential branch; @aburrell - do you feel satisfied with the revised README instructions for Mac installation?

There are a few other issues we should also try to address, including (but not limited to):
5) paths for custom color tables (apologies to @ecbland for breaking this in the first place)
6) `old` vs `new` flags
7) licensing
8) etc

ksterne, 2017-08-30:
Are you thinking of trying for a release a week or so prior to the fall AGU meeting (say Dec. 5th)?  I'm on board with this (if it wasn't obvious from the off-git discussion back in June).  Would you also be in favor of making a `4.1-release` branch about 4 weeks prior to release time (say Nov. 7th) to allow the code to stablize and be better tested?  I haven't looked at it recently, but the merge process should be a lot cleaner this time.

That sounds good on goals for the 4.1 release, is it reasonable that these will be finished by early November?  Do we think 3 would be ready in time for December?  Maybe taking a step back, are goals 1-4 too much of a jump between versions (or more so is too much changing at once)?  Comments on goals (relating to how they were numbered):

1. It seems as though #74 has gotten into some deep discussions on how this code should be implemented and interfaced with dating back to June.  I'll admit I haven't kept up with this one as my focus has been trying to work with #52.

2. Speaking of, it seems as though this is ready to go.  I'm not sure what the hold up is on this.  More testing?  I've tested this a few times and haven't found anything wrong.  This pull request seems to have been ready for a few months though maybe the hold is relating to the median filter discussion dating back to a month ago.

6. It seems as though this is somewhat a broader discussion that might be solved in a working group meeting.

7. I've seen a few attempts at this, but since this is now on github, isn't the licensing mostly obsolete outside of a GNU GPL license?

8. Always love working on etc.

aburrell, 2017-08-30:
@egthomas I'm happy with the README in the `macOS_installation` branch.

egthomas, 2017-09-06:
@ksterne re: licensing - at this point it may be a matter of just picking one and running a script to strip out all of the old license information.  However we handle it, we probably want to wait until one of the "freeze" periods when we know a new pull request won't introduce a merge conflict because it has the wrong license text.

ksterne, 2017-09-06:
Changing the licensing on the release branch could be a path to getting most of the files modified.  However, all of the files wouldn't be able to be modified since any open pull requests would still end up with a merge conflict after the new release is made.  One plan could be a requirement of any new pull request has to have the licensing information changed before it's merged in.  This plus a release branch mass re-licensing could work.

Though doing a lot of work on a release branch can be tricky and difficult.  I thought the problems before were that there wasn't a good way to run a script that would find all of the old licenses (#39).  Has a better script been developed to check this?  It would be tricky though as you'd have to exclude files modified in currently open pull requests.

egthomas, 2017-09-20:
How about we aim for a December release date as @ksterne suggested?  That gives us a couple of months to approve the 3 large outstanding pull requests (fitacf, gridding, map potential).

ksterne, 2017-09-21:
Does that mean we're also good with a release branch creation on Nov. 7th?  That's about 7 weeks from today.  We'd need to get things pulled in before then to make a stable branch before releasing.

ksterne, 2017-10-25:
One thing I forgot to ask during the telecon (sorry it was hard to follow after a while without something visual to look at), is there something in particular that will/wants to make the release branch?  Should I do it?

kkotyk, 2017-10-25:
I think whatever you did last time worked well. I would be okay with you making it again.

ksterne, 2017-11-06:
Comments?  Are we ready to go for tomorrow?  Should I make a release branch tomorrow?

pasha-ponomarenko, 2017-11-06:
Keith is just checking up with Evan with respect to keeping the new elevation algorithm optional in FITACF2.5, as we agreed upon at the telecon.

sshepherd, 2017-11-06:
I was not present on the telecon but Evan debriefed me. I am not willing to
spend any more time on the RST, so if someone else wants to do whatever was
agreed upon, I can't stop you.
Questions about the status of the algorithm will be redirected to the WG
chair.

It's quite impressive that the bar to get algorithms accepted into the RST
is now higher than getting a publication of said algorithm accepted in an
AGU journal! Well done.

Simon

egthomas, 2017-11-07:
I'm not sure that we should be opening a release branch today if none of the major pull requests have been merged yet.

ksterne, 2017-11-07:
Sorry to hear that you won't be able to help us out @sshepherd, we're in need of more people to test and/or develop.

I agree @egthomas, lets see if we can get #52, #100, and #74 finished up this week.

ksterne, 2017-11-17:
I think we've now accomplished goals 1-4.  So with that I'll close this "issue" and create a release.4.1 branch.

closed 2017-11-17 by ksterne