SuperDARN/rst #101: Default options for time_plot routine

ecbland, 2017-09-25:
@egthomas and @pasha-ponomarenko suggested that we change some of the default options in `time_plot` (see #99). 

The following changes have already been suggested (feel free to suggest more in the comments): 

1. Plot beam 7 by default instead of all beams (all beams can then be plotted by setting `-b -1`)
2. Change the default velocity scale to +/- 1000 m/s instead of +/- 2000 m/s
3. Choose a new colour table for the velocity

I have worked out how to implement all of the above changes. (1) and (2) are trivial. (3) is moderately straightforward but we first need to agree on what the new colour table should be!

@pasha-ponomarenko suggested that we use IDL color table 4 (see http://www.harrisgeospatial.com/docs/LoadingDefaultColorTables.html). This works well on a white background and has a smooth transition from positive to negative velocities. However, it doesn't work well on a **black** background, which is also an option in `time_plot`.

So, please make suggestions! We can create any colour table we like, so there's no need to feel restricted to existing colour tables from IDL, for example. Personally, I prefer the table used on the VT website (red-orange-green-blue). 

Note that any changes to the velocity colour table in `time_plot` will be applied to the other plotting routines in RST as well.

sshepherd, 2017-09-25:
> LoadingDefaultColorTables.html). This works well on a white background

My preference is a simple blue (toward) and red (away) scale. color table 4
looks "pukey" and the VT scale is confusing and emphasizes the low velocity
greens.

kkotyk, 2017-09-25:
We had a brief discussion about color scales here at Usask at one point and someone brought up the point of being color blind friendly. I'm not sure how much that is needed, but it would certainly be appreciated by some people out there I imagine.

pasha-ponomarenko, 2017-09-25:
The most widespread type of colour-blindness is that a person does not distinguish between red and green. This, besides estetic considerations, rules out my favourite table.
:-(
The red-and-blue one is good because it si intuitive. The only problem is that if one mixes it directly, the intermediate colours are very bright purple.
[image]
An alternative might be to merge them through a shade of grey so that it would also mark most of the ground scatter as grey, according to the current unwritten convention
[image]
I am pretty sure a saw one of those somewhere... It is possible to play around with ajusting how quickly the merging occurs (in the above case it is linear).
It is very easy to generate this sort of table in IDL, but I don't know how hard it is to do this in C...

egthomas, 2017-09-25:
I don't have a major issue with the default velocity color table used in `time_plot`, although I do agree it probably emphasizes the transition between low positive and negative velocities too heavily.  Is it worth broadening the scope of this issue to consider the default parameter scales for some of the other plotting routines as well?  For example, do we need to make similar adjustments to the default velocity scale in the `field_plot` routine?

pasha-ponomarenko, 2017-09-25:
@egthomas, another problem with the current table is that the maximum positive velocities merge into the white background.

ecbland, 2017-09-25:
@egthomas:

> Is it worth broadening the scope of this issue to consider the default parameter scales for some of the other plotting routines as well? 

Yes, I was thinking about that too. I agree.

pasha-ponomarenko, 2017-09-25:
I would also reconsider the way the noise level and transmiter frequency are plotted at the top of the range-time plot. Conventional plots (like those from the VT webpage) would be easier to interpret plus the respective colour bars will disappear. If we will decide to go this way, than I would propose using dots rather than lines.

ecbland, 2017-09-26:
@pasha-ponomarenko 

> It is possible to play around with adjusting how quickly the merging occurs (in the above case it is linear).

Yes, yes it is! A few options are shown below. I also experimented with the particular shade of grey (RGB={128,128,128} and RGB={169,169,169}). 

[image]
[image]
[image]
[image]

egthomas, 2017-09-26:
`time_plot` already has a flag to shade ground scatter gray, right?  Do we even need to manually insert it into the default color table?

(@ecbland - sorry, I missed the last sentence of your original message the first time around - you're already on top of things!)

ecbland, 2017-09-27:
@egthomas:

> time_plot already has a flag to shade ground scatter gray, right? Do we even need to manually insert it into the default color table?

Yes, the ground scatter can be shaded grey using the `-gs` option. However, if we use the proposed red-blue colour table for the velocity, we still have to put another colour in the middle or else we get purple at 0 m/s, which is hard to read (see Pasha's plots above: https://github.com/SuperDARN/rst/issues/101#issuecomment-331947713). The grey is then an obvious choice because it's neutral, and will show up on both white and black backgrounds. 

Whatever we decide, I'm sure there will people who would prefer to use something else. We could add some environment variables to `.profile/superdarn.bash` which allow the user to set their own default colour tables, i.e.

- `export DEFAULT_VELOCITY_COLOR_KEY=${COLOR_TABLE_PATH}"/[name.key]"`
- `export DEFAULT_POWER_COLOR_KEY=${COLOR_TABLE_PATH}"/[name.key]"`

ecbland, 2017-11-07:
During the working group meeting on 25 October we agreed that this issue is of low priority, so I'm closing it. 

In case anyone would like to revisit this in the future, the relevant source code for changing the default color keys is in `/rst/codebase/general/src.lib/stdkey.1.5/src/make_palette.c`, in particular the routines `make_reg()` and `make_root()`

closed 2017-11-07 by ecbland