SuperDARN/rst #103: How are the IDL DLMs used?

kkotyk, 2017-10-03:
Just wondering how the IDL DLMs are used. I'm not really sure of their relationship to the the startup.pro file. Maybe you know @egthomas? Thanks.

pasha-ponomarenko, 2017-10-03:
@kkotyk 

I am not sure if this is going to help, but there are some descriptions here

http://www.harrisgeospatial.com/docs/DLM.html

and here

http://northstar-www.dartmouth.edu/doc/idl/html_6.2/Dynamically_Loadable_Modules.html

egthomas, 2017-10-03:
@pasha-ponomarenko beat me to it.  Basically, DLMs are just wrappers for C functions.  In the RST we have DLMs for opening/reading the various file formats, performing coordinate transformations, and loading the radar table information.  Sometimes it can cause confusion because we have equivalently named IDL and C functions and it is not always obvious which is actually being used from within IDL.

kkotyk, 2017-10-03:
Okay thanks.

closed 2017-10-03 by kkotyk