SuperDARN/rst #114: missing hdw.dat.svb

asreimer, 2017-12-07:
The `hdw.dat` file for SPEAR is missing from the this repository.

If you look at [the radars.dat file](https://github.com/SuperDARN/rst/blob/master/tables/superdarn/radar.dat) it references SPEAR.

However, [the tables/hdw  directory](https://github.com/SuperDARN/rst/tree/master/tables/superdarn/hdw) doesn't have the hdw.dat.svb file.

This issue also occurs for the [VT hdw.dat repo](https://github.com/vtsuperdarn/hdw.dat). [I've also filed an issue there.](https://github.com/vtsuperdarn/hdw.dat/issues/5)

egthomas, 2017-12-07:
Huh, didn't realize there was a hdw.dat file for SPEAR in pre-VTRST3.5 releases.  I'll create a pull request for a copy we have from version 3.3 of the RST.

asreimer, 2017-12-07:
Frankly, I was as surprised as you. Thanks for fixing this.

egthomas, 2017-12-07:
I think the brief but spirited discussion in the closed #115 pull request also brings this issue to a close - the `hdw.dat.svb` file will not be returned to the RST and the SPEAR entry has been removed from the `radar.dat` file on the 4.1 release branch.

closed 2017-12-07 by egthomas