SuperDARN/rst #116: test_raw and dattorawacf buffer overflow with 1993-1995 dat files

egthomas, 2017-12-07:
I don't have much time to look into this with AGU approaching but I'm consistently encountering buffer overflows when trying to process early (~1993-1995) dat files using the `test_raw` and `dattorawacf` binaries.  I think it might be happening in `OldRawRead` of the `oldraw.1.16` library although am not entirely sure.  Is anyone else experiencing these issues?

A few of the files I've tested include (obtained from the BAS data mirror):

1993093023t.dat
1993101221g.dat
1993101221k.dat
1993101221t.dat
1994010100g.dat
1994010100k.dat
1994010100t.dat
1995010100k.dat
1995010100w.dat

pasha-ponomarenko, 2017-12-07:
I did recall a problem related to reading/converting dat files in the past, but it looked more like a segfault. Is it only for the 1993-1995 data?

egthomas, 2017-12-08:
The half dozen dat files I tried from the year 2000 worked without any buffer overflow errors.  Beyond that I haven't narrowed down a specific problem interval.  It seems like the routines make it through all of the records in the file but crash on the next iteration of the loop (ie when it should hit a null or something and exit).

egthomas, 2018-01-16:
I think I've tracked this down to the `int rlen` parameter in the OldRawFp structure:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/oldraw.1.16/include/oldrawread.h#L45

Sometimes OldRawRead appears to figure out that the record length at the end of the file is zero and not crash (post-1995 .dat files?) but other times it doesn't realize the file is about to end and you encounter a buffer overflow (~1993-1995 .dat files).  I tested this by using the IDL OldRawOpen and OldRawRead routines (without DLMs) and checking the rechdr.size (or rawfp.rlen) values to see if they are zero.  Sure enough, the 1993-95 files appear to have "empty" records at the end of the .dat files - the file pointer doesn't seem to return a null but the record header information is all zero / missing so it crashes when trying to access the full set of `(struct OldRawFp *ptr,struct RadarParm *,struct RawData *)` here:

https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/oldraw.1.16/src/rawread.c#L49

Hopefully that makes enough sense for a better programmer than me to fix.  To summarize, the output from `dattorawacf` for these early files is in fact complete despite the buffer overflow.  You can do a `dattorawacf`, `rawacftodat`, and `dattorawacf` to complete the circle and see that a buffer overflow does not happen on the final `dattorawacf` call.

egthomas, 2018-01-22:
@kkotyk - any thoughts on this?

kkotyk, 2018-01-22:
Despite my joy of sifting through DMAP, I haven't had a chance to do the same for the dat format. From your investigation it seems like its a similar issue to when dmap has a zero in the size field. I've also encountered some issues that makes me question dattorawacf. I think its worth it in the future to investigate that utility if someone gets a chance. I noticed that many of the corrupt files rawacf files I looked at were processed with dattorawacf.

egthomas, 2018-01-23:
I guess that raises any interesting question - how do people usually deal with data originating from .dat files?  Do they go `dat -> rawacf -> fitacf`, or `dat -> fit -> fitacf`, or just `dat -> fit` ?  This seems pretty important since the first 12+ years of digital SuperDARN data are in .dat format.

kkotyk, 2018-01-23:
This is a good question. One of the things I was discussing with Pasha is, should make_fit just always provide fitacf DMAP output regardless of input? I know I usually convert dat to rawacf and then use my python code to error check, but I cant say anything about the actual data itself.

egthomas, 2018-01-25:
I think in the short term we also need to update the documentation on `make_fit` for old format files.  When you pass it a rawacf file, you can use the usual syntax such as

`make_fit 20180101.sas.rawacf > 20180101.sas.fitacf`

but with the older dat files the syntax is

`make_fit -old 20180101.sas.dat 20180101.sas.fit` (ie output file is not written to stdout).  Here is the relevant documentation page:

https://superdarn.github.io/rst/superdarn/src.bin/tk/tool/make_fit/index.html

That was likely my mistake when switching the old and new flags on make_fit.

kkotyk, 2018-01-25:
Yeah thats a good idea. I'm pretty sure make_fit has always worked that way and I dont know why it was designed with 2 different invocations based off input type.

asreimer, 2018-01-25:
@egthomas 
> I guess that raises any interesting question - how do people usually deal with data originating from .dat files?

If it means anything to you guys, when I always converted from `.dat` to `.rawacf` and then performed fitting. It doesn't make sense to me to process using `fit` and then convert to `fitacf`. Process all the data the same way.

In the interests of operational processing and maintaining a database (I maintain 150TB of incoherent scatter radar data now), it is much easier to maintain as much data as possible in the `latest` format.

Is it even useful to keep `fit` files? Does anyone use them? More importantly, **should** anyone use them?

egthomas, 2018-03-26:
Just to clarify, the file in question is actually `rawreadcurrent.c` in the oldraw lib instead of `readrawcurrent.c`.

I see a couple of similar checks against the output of ConvertReadShort() in various cmpraw files and OldRawOpenFd of rawopen.c in the oldrawlib - should those be changed too?

sshepherd, 2018-03-26:
Wow... That's the most obvious example of dyslexia I've ever seen... Aaah, good catch Evan!

The others probably warrant a careful look. The problem was asking to read a negative number of bytes. If that is happening elsewhere, then you should get the same behavior. I believe `OldRawOpenFd` is called by `dattorawacf`, which seems to be working well on .dat files from 1993-1998 (haven't checked the others yet.)

sshepherd, 2018-03-26:
@egthomas is correct. That same issue appears in `OldRawOpenFd`.

egthomas, 2018-03-28:
I've verified that @sshepherd's fix to the C code works and created a pull request also implementing the fix in IDL (although I think you only need to check for a zero rather than negative number of bytes).

sshepherd, 2018-03-28:
I actually think the modification can be made simpler, i.e.,

`+     if (ConvertReadShort(fp->rawfp,&num_byte) != 0 || num_byte <= 0) {`

There is no performance penalty because of the short-circuit evaluation and I have changed back to the original `<=` instead of the `==`. While the latter may be sufficient, the former is "safer" in that you would not want `num_bytes` ever to be negative, which it could be in the case of something causing it to be returned in an uninitialized state. I expect there is something at the end of some of the files that is causing the call to `ConvertReadShort()` to return a success, but there is nothing past that to read. In most cases `num_byte` appears to be set to zero, but I would suggest the `<=` check to be safe.

I have processed the entire set of .dat from the BAS server and have not, to my knowledge, had a problem with any of these files.

I am happy to attempt the change, which would be a good exercise for me to go through after being away from git for several months...

egthomas, 2018-03-28:
@sshepherd - yes, that implementation seems to work too.  Do you want to add it as a commit to the open pull request (#139)?  I'm happy to handle it too since I was just looking at some of the other functions we'll need to modify.

egthomas, 2018-03-28:
Disturbingly, the `cmpraw` output is showing a lot of differences in the acfd array between an original dat file and a supposedly equivalent dat file produced by going `dattorawacf` -> `rawacftodat`.  Can anyone else replicate this?

sshepherd, 2018-03-28:
Perhaps this is a larger issue that relates to .dat files. I cannot remember the issues associated with going from one form to the other, but recall that it was not entirely trivial. Perhaps someone like Dieter would remember...

There are other issues with .dat files that might spill over into the data distribution WG. For instance, the following files:

```
2003041420b.dat
2003041420ba.dat
2003041420bb.dat
```

should contain sequential timestamps, however, these three files contain records with the following timestamps:

```
**2003041420b.dat**
2000:00
2000:03
1915:12
1915:15
...
1915:53

**2003041420ba.dat**
2000:00
2000:03
2000:06
...
2013:22

**2003041420bb.dat**
2059:32
2014:43
2014:46
...
2058:19

```
So records are not always sequential within a given file or between files, which I believe they should be.

There are also some .dat files from 2006 that are seg faulting when doing `dattorawacf`, e.g., `2006012208b.dat`.

pasha-ponomarenko, 2018-03-28:
I just tested 2001060502f.dat and obtained ~64,000 different readings with the there-and-back file being smaller in size. I am not sure which stage this happens. It would make sense to compare the actual arrays in all three files in order to find out at which leg the discrepancy occurs.

pasha-ponomarenko, 2018-03-28:
I just read the raw structures into IDL. The discrepancy occurs at the first leg while converting from dat to raw. Some of the non-zero ACF and XCF values are turned into zeroes at this stage. No discrepancies arise from the inverse transform.

egthomas, 2018-03-28:
Could it have something to do with this "threshold" parameter you can set as a command line option for `dattorawacf` and `rawacftodat`?  

https://github.com/SuperDARN/rst/blob/e6e9a1a0427436555dc84eea1e374becf0f59921/codebase/superdarn/src.lib/tk/oldraw.1.16/src/rawwrite.c#L248-L274

It looks like values are not necessarily written depending on the lag0 power?

pasha-ponomarenko, 2018-03-28:
I have just run it with zero threshold. It generated a file which is bigger than the original dat file, but the contents (acfd, xcfd and pwr0) seem to be identical now. 
It is a strange way to introduce a threshold: instead of the default value being zero it is set to the half of the noise level.

egthomas, 2018-03-28:
Is that threshold value something that's typically set at the control program level, or in the radar OS, or what?  I can see where it's used in ACFSumPower but that's only called when making a raw file from an iqdat or in site.c on the ROS side of things, so I assume it's determined on a per-radar basis.

Is there any reason users should be able to set this value when they go from dat to rawacf?  Have we changed the way we use/set this threshold value over the course of SD's history?

pasha-ponomarenko, 2018-03-28:
It is the first time I came across this thing. In my opinion, the OS/CP issues/thresholds should not affect this kind of a post-measurement type conversion anyways. Too much for data replicability!

With repsect to the use of the on-site threshold, one needs to contact people familiar with on-site operational software. I can only recall observing that some .dat files from HAN would have all pwr0 values but some of the respective low-power ACFs below a certain threshold would not be recorded. I have even worked out the threshold value for one of the records, and it was not based on the 10 weakest lag 0 power values used in FITACF1-2.

egthomas, 2018-06-29:
Closing due to #139 being merged.

closed 2018-06-29 by egthomas