SuperDARN/rst #117: Archiving old branches as tags

egthomas, 2017-12-09:
@kkotyk and I discussed this briefly via email, but I've started to "archive" two outdated/unused branches (`cs10` and `master-old`) using the tag approach described at the links below:

https://stackoverflow.com/questions/1307114/how-can-i-archive-git-branches
http://www.aaronwest.net/blog/index.cfm/2011/6/7/Git-Workflows-Archiving-Old-Branches

This means the branches can be safely deleted to reduce clutter in the branch list but can be fully restored from the tag if necessary.  I haven't yet deleted the two branches in question, only tagged them. 
 How do other people feel about this approach?

mtwalach, 2017-12-11:
Sounds like a good idea.

ksterne, 2017-12-11:
It seems like a good idea.  I thought `master-old` was a way to store the previous version master release, but using a tag is probably better.

closed 2017-12-20 by egthomas