SuperDARN/rst #119: make_grid '-c' flag and mixing radar data

billetd, 2017-12-15:
Whilst using the -c flag in make_grid, I've noticed that it won't concatenate data from different radars. I initially thought it was a shortcut so combine_grid didn't need to be used, but I see that '-c' only works when concatenating from the same radar at different times. This isn't mentioned in the make_grid documentation (https://goo.gl/UwMZiq), but is this the intended behaviour?

egthomas, 2017-12-16:
Yes that is the intended behavior that make_grid only operates on data from a single radar at a time, even with the -c flag set for handling multiple input files.  We can certainly update the documentation to clarify that.

billetd, 2017-12-18:
Thanks for the reply Evan. Yes it would be good for this to be in documentation. If you use make_grid with multiple files from different radars, it appears to work on them normally without trouble, but the created grid file only ends up with the first radar in the list. Thank you for clarifying!

egthomas, 2017-12-19:
No problem!  I've just updated the command line documentation on the 4.1 release branch as well as on the gh-pages branch for the online documentation.  Is it safe to close this issue?

billetd, 2017-12-19:
Great thanks!

closed 2017-12-19 by billetd