SuperDARN/rst #127: rst/codebase/superdarn/src.idl/lib/legacy.1.6/global.pro - missing procedure definition

mtwalach, 2018-02-01:
I have noticed that [global.pro](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.idl/lib/legacy.1.6/global.pro) is missing a definition at the beginning.

Is it supposed to be `FUNCTION global`?

I don't use it, so I don't know what it's for or what it's supposed to be, but anyone who tries to collectively compile the idl libraries using [startup.pro](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.idl/startup/startup.pro) will run into a `End of file encountered before end of program.`

To confuse the matter even more, the comments at the beginning or `global.pro` suggest the file is supposed to be `aacgmlib.pro`, but that's a separate file in the [legacy folder](https://github.com/SuperDARN/rst/tree/master/codebase/superdarn/src.idl/lib/legacy.1.6).

egthomas, 2018-02-01:
@mtwalach the same thing happens to me when I try calling `@startup` from within a different  .pro file pointed to by `IDL_STARTUP` (similar to http://davit1.ece.vt.edu/doc/startup/davit_init.pro).  Therefore I've also been commenting out the `.run global.pro` line from my local `startup.pro`.

If someone keeps their `IDL_STARTUP` environment variable set to the RST default of `"${RSTPATH}/idl/startup.pro"`, then adding a `pro global` or `function global` definition to the beginning of `global.pro` seems to prevent the values written to the variables/common blocks from being stored in memory.  For example, if I keep all of the RST defaults then this is what I get when running IDL from the command line:

```
...
% Compiled module: ISTP_CDF_HELP.
% Compiled module: ISTP_CDF.
% Compiled module: ISTP_CNV_GSE_GSM.
% Compiled module: ISTP_CNV_GSM_GSE.
% Compiled module: $MAIN$.
IDL> help,true
TRUE            INT       =        1
IDL> help,false
FALSE           INT       =        0
IDL> help,parms_size
PARMS_SIZE      INT       =       48
```

But if I add a procedure or function definition to the beginning of `global.pro` then this is what I get:

```
% Compiled module: ISTP_CDF_HELP.
% Compiled module: ISTP_CDF.
% Compiled module: ISTP_CNV_GSE_GSM.
% Compiled module: ISTP_CNV_GSM_GSE.
% Compiled module: GLOBAL.
IDL> help,true
TRUE            UNDEFINED = <Undefined>
IDL> help,false
FALSE           UNDEFINED = <Undefined>
IDL> help,parmse_size
PARMSE_SIZE     UNDEFINED = <Undefined>
```

So I think adding that definition would prevent someone from using the values being set within `global.pro`.  Although that begs the question of whether anyone still relies on these variables being set?  The "legacy" directory name seems at least partially inappropriate since some very useful routines like cnvcoord, mlt, and rbpos are located there.

egthomas, 2018-02-01:
I just created an `idl_cleanup` branch fixing the aacgmlib/global naming confusion and a few other minor errors - feel free to add to it!

egthomas, 2018-03-30:
Closing this issue since #128 was merged.

closed 2018-03-30 by egthomas