SuperDARN/rst #148: Negative scan flag

mts299, 2018-04-24:
Currently, the scan flag -1 is being handled by taking the absolute value done in the pull request #132 in the develop branch. 

This issue is to address other possible situations where the negative scan flag can indicate not to use that record of data and discuss possible fixes or how it should be addressed in the code. 

A further discussion is planned to be done at the 2018 SuperDARN workshop.

mts299, 2018-06-20:
Did we figure out the verdict at the conference?

egthomas, 2018-06-20:
Mark said he would ask Steve and Tim about this.  Maybe @arfogg can follow up with Mark?

arfogg, 2018-06-21:
Hi,

Steve's definitions of the scan flag are:

; Use fit file scan marks to create scans:
; scan_mark() = +/- 1  - start of a new scan
;             = 0      - include in current scan
;             = -32768 - camp beam - don't include in scan

Taken from PRO make_scans in go_read_routines.pro in Go.

He also said that there were some instances (a very small subset of the data) from the mid 90s where the scan flag was incorrectly set in some fit files.

Overall: we don't think there are instances where scan flag=-1 is indicating not to use the data.

Hope this helps, Alexandra.

ksterne, 2018-06-21:
I thought some of this would be covered in the reference document I was talking about where we would define all of variables and with things like scan flag, what are the possible values.

egthomas, 2018-06-25:
@arfogg thanks Alexandra.

@ksterne I agree that should be explained somewhere, but it seems too technical for the "RST for dummies" guide @ecbland and I were supposed to put together in the near future.  For now I'm going to close this issue but we can always refer back to it in the future.

closed 2018-06-25 by egthomas

ksterne, 2018-06-26:
@egthomas, I think it's good to close it here since we've been talking about ways to document this.  I thought there were a few different documents we'd discussed at the Workshop that would be targeted at different audiences.  One of which was the "RST for dummies" that I think Kathryn requested so that once you've got the RST software package installed, you know what to do with it.  The other document was one I was referring to that is more of reference document describing all of the parameters in all of our files and defining possible values.  The scan flag being one of these.  From reviewing the minutes, I guess I volunteered to head that document, so maybe I should get on that soon...