SuperDARN/rst #160: Spectral width w_l negative?

ksterne, 2018-05-24:
Hey all, this is probably one of those things where it would be good to have a guide describing what each of the variables means in each of our data products.  Maybe one exists, and if so, maybe someone can point me to it.  But our students were working here and noticed that there's a significant amount of spectral width (Lorentzian fit), w_l, values in fitacf files that are negative.  Is this something that's supposed to be and if so, can someone give some meaning to why a spectral width would be negative and not positive?  Does it have something to do with signals being reversed from some expected response in the frequency domain?

pasha-ponomarenko, 2018-05-24:
@ksterne, this is a well-know issue with the exponential fit. Just a few weejks ago Evan asked the same question. The bulk of the negative spectral widths occurs in the ground scatter for which the power vs lag dependence is very flat so that due to fitting errors it might be either slightly begative or slightly positive. It just means that the width estimate is of the same order as its fitting error.

pasha-ponomarenko, 2018-05-24:
There should be " its fitted slope might be " instead of "it might be".

pasha-ponomarenko, 2018-05-24:
The only reference to the fitted data products which I am aware of is the old Rob's description (I copied it as PDF and stored here  at https://homepage.usask.ca/~pbp672/DAWG.html in "Supplementary materials" as "FIT data format"

egthomas, 2018-05-24:
@pasha-ponomarenko the source for that page is actually included in the RST; the html version can be viewed here:  https://superdarn.github.io/rst/superdarn/src.doc/rfc/0009.html

pasha-ponomarenko, 2018-05-25:
Thanks, Evan, I was pretty sure it is somewhere there!

ksterne, 2018-05-25:
Mmmm, we'll that's a start I guess.  Though it would be good to put a little bit more onto the description.  What does positive/negative convention mean?  Are there limits like must always be positive?  Or any error codes?  One might even have some fun trying to define "scan" ;).

sshepherd, 2018-05-25:
A scan is just a sequence of beam soundings that repeat.

pasha-ponomarenko, 2018-05-25:
I thiought I had Kile's tutorial from early 2000s, but apparently I hadn't, so I added it to the "Supplementary materials" list. Here is its direct link:
https://homepage.usask.ca/~pbp672/Kile.pdf

closed 2019-04-10 by egthomas