SuperDARN/rst #163: Which version of RST?

mtwalach, 2018-06-15:
Is there an easy way to find out which version of/particular RST branch someone is running (something that works cross platform)? Obviously, if I pull directly from GitHub, I can look at my log, but it won't tell me if I actually re-installed it or just pulled it.. I was also wondering if there is something that works for people who don't git.

If this doesn't exist, can we create some kind of help keyword like "rst --help" that will tell you? 
I think this could be quite useful, especially if people reinstall a newer version or branch to test if it's been done properly or if they only use process data occasionally and can't remember what they installed or when they last updated it..

Thanks!

mtwalach, 2018-06-16:
> Is it easy to add the RST version to the output of RST programs?

That's exactly what I was thinking.. That way, even if you have several copies/versions downloaded, but only one installed you can easily find out which one it is.

egthomas, 2018-06-17:
@asreimer we don't manually rename the directory when we create a new release; that information is taken from the tag we assign and handled by github.  We also haven't been updating the library/binary version numbers because it makes the commit history difficult to follow.  I tried experimenting with the `git mv` command but wasn't terribly impressed with the result.  

I do like the idea of putting a version file in the top-level RST directory.  How would we handle the case of between-releases?  For example, would the current develop branch retain the 4.1 version number?  Or should it say something like 4.1-develop?  Or 4.2-develop?  Or 4.2-beta?  I like the output of git describe because it indicates which version the branch is based on and how many commits have been made since the release, e.g. `v4.1-67-gf72306a`.  But again that isn't very useful for non-git users.

egthomas, 2018-08-03:
Addressed in #173

closed 2018-08-03 by egthomas