SuperDARN/rst #168: Installing RST as a superuser

mtwalach, 2018-07-12:
Hi,

I have been trying to install RST on a Linux system (Debian) as a normal user with root privileges, but if I run make.build using sudo it fails miserably as it can't find any of the paths that should be pointed to by "$RSTPATH/.profile.bash".
I don't know if this is system dependent or if my Linux knowledge doesn't stretch far enough, but I think it all boils down to the fact that you can't do "sudo source ~/.bashrc" and if you just do "source ~/.bashrc" it doesn't translate to the superuser.

This is probably not generally a problem for people, but is there an easy way around this? 

I don't know if anyone has encountered this before, but if there is a simple solution that I have not thought of I'd be interested (and maybe it should accompany the install instructions).

Thanks!

mts299, 2018-07-12:
@mtwalach - why are you installing RST with sudo?
Did you try without using sudo?

Also, you cannot use sudo with source because of how bash scripts are executed regarding sub-shells vs. shells and who owns that shell session. When you run a script you create a sub-shell that runs all those commands, thus any environment variables you set in those scripts will not be saved (which is a bug in davitpy's build scripts). However, by running source on a script, it evaluates the script in the current shell (terminal) for that user. When you use sudo with source the command gets confused as to which user owns the shell thus the command does not work. 

If you want to run your bashrc in root, then you have to sudo into root and then run source on the script. So if the problem is more about how to install RST at a root level, then you just need to sign into root and install it from there, just be careful with privileges if you want other users to be able to use it. 

Hopefully, this answers your questions :)

mtwalach, 2018-07-16:
Hi Marina,

Yes, the problem is that I (as a normal user) don't have the right permissions, which is why I thought I'd try using sudo. We had a change of staff recently and the person who used to be admin on that machine has left, which is why using sudo seemed the simplest option. 

I didn't know if this was something obvious, but it makes more sense now. Thanks for explaining that. 
We'll explore the root option - thank you!

mtwalach, 2018-07-19:
We solved it in the end by using sudo interactively ("sudo -i")!

closed 2018-07-19 by mtwalach