SuperDARN/rst #170: Following build instructions results in tar error

brianbienvenu, 2018-07-19:
Greetings from Australia!

I'm following through the build instructions on the root README.md:
https://github.com/SuperDARN/rst/blob/e6e9a1a0427436555dc84eea1e374becf0f59921/README.md

Got down to 4a - my build is saying:
```
Failed to locate idl_export.h - will not compile IDL libraries or DLMs
```
Running the suggested fix results in an error:
```
/rst/codebase/superdarn/src.lib/tk# tar -P -czuf idl.tar.gz idl
tar: You may not specify more than one '-Acdtrux', '--delete' or  '--test-label' option
Try 'tar --help' or 'tar --usage' for more information.
```

I see it has both the `-c` (create) and `-u` (update) operation modes specified, which is the cause of the error.

Does anyone know what this snippet is actually trying to do? `idl.tar.gz` already exists in that directory in the repo.

brianbienvenu, 2018-07-19:
Hi Evan. I'm installing on a fresh Ubuntu 16.04 - no IDL.

The build originally wasn't continuing, because I was rather stupidly calling `make.code superdarn rst` - muscle memory from an old version. As a result, I assumed that this error message was an actual problem.

Running `make.code` by itself (as per the README today) works as intended.

Regardless, the instructions in step 4a appear to be in error, as they have both `-c` and `-u` flags set, which `tar` will complain about.

egthomas, 2018-07-23:
Hi Brian - I think you're right that the readme instructions should say `tar -P -czvf` instead of `tar -P -czuf`.  It is possible that was a typo no one else caught.

egthomas, 2018-08-01:
The readme tar instructions should now be fixed and will make it into users' hands once we agree on the next official release.

closed 2018-08-01 by egthomas