SuperDARN/rst #174: 4.2 release?

sshepherd, 2018-07-26:
What is the plan for releasing 4.2? It has been a long time since anything new has been released. There are known bugs in the current version and users are circumventing github to get updates. AACGM_v2 is one example. We need a mechanism that allows for a much quicker turn-around on bug fixes. Our users will be testers, but not if the bug fixes sit in the RST for months or years.

v4.2 seems quite extensive so perhaps a v4.1.1 release would be better. It would get the important (or already "tested") changes out to users and simplify the develop branch.

egthomas, 2018-07-26:
I agree there have been a lot of bug fixes since the previous release over 6 months ago - many of which have been the direct result of user feedback.  Personally, I would still like to see several more fixes approved before even a 4.1.1 bug-fix-style branch is released (eg, #133,  #151,  #152, #153, #171, #172) but I understand that the line needs to be drawn somewhere.

pasha-ponomarenko, 2018-07-26:
Let me fend off the reviewers of my tdiff paper, and we will have a go at this (sub)release issue...

ksterne, 2018-07-31:
@sshepherd, are you thinking of some kind of (rough/loose) schedule for making releases (even if they're patch releases) in order to propagate new features to the `master` branch?  I think we were trying to lay that out in the release guidelines document that Evan and I drew up, but I don't remember any review/acceptance of that for this repo.  I think having an agreed upon document for making new releases would help guide drawing the line for us.

@sshepherd, are there any particularly critical bug fixes (aka hotfix) that you can think of that aren't in the `master` branch?  I think we've been pretty good on getting hotfixes into `master`, but I agree there's a need to get new features from `develop` to `master` for a new release as we have had a new release in a while.

ksterne, 2018-07-31:
Diving a little deeper, minutes from DAWG meeting at SD '18 note a need to distribute the release guideline document as most people hadn't reviewed it and so it couldn't be accepted or reviewed.  @pasha-ponomarenko, do you want to take this on?

sshepherd, 2018-07-31:
I've completely lost track of where things are with the RST but do know
that there are bugs in AACGM-v2 that have been fixed, perhaps still waiting
for testing (I don't know) but the bugs are still in RST 4.1, which people
are downloading. There are many other things I can think of (TS18
statistical model, elevation angle algorithm, etc.) that would be useful to
get into the hands of users. There seems to be a lot of hand-wringing and
planning for a release, but nothing much is actually happening. I feel that
it would be much more useful to get these changes to users. If someone
(tester or user) asks me a question about something that I put in to github
a year ago, I will have nothing to say since I can barely remember what I
did last week.

pasha-ponomarenko, 2018-07-31:
@ksterne, yep, it has to be distributed. How can I get the last draft?

ksterne, 2018-08-10:
Now we've got the release guidelines fairly well accepted (I think) bringing things back here.  It seems like only #133 and #152 are on the original list @egthomas put forward of things to be merged.  It seems like these are both fairly imminent, so should we create a `4.2\release` branch from `develop` once those are merged in?  If there are other pull requests, I think those might better go into a 4.2.1 release.

egthomas, 2018-08-10:
If @mts299 is comfortable with merging the `plotting_fixes` branch then I think we will be ready to create a `4.2.0-release` branch.  The only caveat is the Mac environment variable issue raised by Adrian just within the last day or so and described in #77 - help from any Mac users would be greatly appreciated.

pasha-ponomarenko, 2018-08-10:
Marina wrote yesterday

> I think plotting_fixes is ready to be merged if you want to include the PR in the 4.1.1 release. 
It is "bug fixes" which is part of minor releases. 

so I will just merge 152.

ksterne, 2018-08-10:
Should I go ahead and make the new release branch?  Or does someone else feel up for it?  Maybe I can while on layovers tomorrow.

ksterne, 2018-08-11:
Think is is now resolved.  Mind closing it @sshepherd?

sshepherd, 2018-08-11:
I would but for the fact that I have limited connectivity here in Timmins.

closed 2018-08-11 by sshepherd