SuperDARN/rst #177: Which IQDAT dmap fields are actually used?

brianbienvenu, 2018-08-08:
Hello!

I'm presently implementing a new "operating system" named Mabel for controlling TIGER-3 and have gotten to the pointy end of the stick with IQDAT file generation.

My plans, in brief:

- Control: Python process, handling scheduling, all setup, UDP comms with FPGAs, etc. Passes received UDP packets to next via Python multiprocessing.Pipe.
- Processing: Python process, coalesces data from packets, does whatever formatting required to write IQDAT file.
- RAWACF: Pipe in from IQDAT above
- FITACF: Pipe in from RAWACF above

I see **many** fields in the IQDAT format: https://github.com/SuperDARNCanada/backscatter/blob/master/backscatter/dmap/dmap.py#L1308

For most of them, I can interpret their use (especially with help from [A. Reimer's thesis](https://ecommons.usask.ca/bitstream/handle/10388/8534/REIMER-DISSERTATION-2018.pdf) and cmpfit's [raw.h](https://github.com/SuperDARN/rst/blob/e6e9a1a0427436555dc84eea1e374becf0f59921/codebase/superdarn/src.bin/tk/testing/cmpfit.1.6/raw.h)). Some of them seem to be just informational, some quite redundant.

#### Informational

- cp
- ercod
- intt.sc
- intt.us
- mplgs
- mxpwr
- noise.mean
- noise.search
- origin.code
- rxrise
- scan
- stat.agc
- stat.lopwr

#### Redundancies

- time.*, tsc[], tus[]
- rsep, smsep (overconstrained by physics - smsep*c/2 = rsep)
- lagfr, frang (overconstrained by physics or rsep/smsep)
- toff, tsze (size can be inferred from smpnum, offset can trivially be found from multiplication of tsze)
- ltab, can be inferred from ptab

#### Random numbers in old data

- badtr (bad transmission times in microseconds? - related to bad lags?) Length in btnum.

### My query
Does anybody know which IQDAT fields are used during the make_raw (or other downstream) processing, and which ones I can leave out? Can I remove all of the *informational* fields above?

Does anybody know of anything using badtr in IQDAT files?

ksterne, 2018-08-08:
In addition to that, it might be helpful to ping the darn-oswg list as they are ones that are concerned with radar operating software.  This repo has largely stripped out code related to radar operations as it's more focused on data analysis/processing.

I would say though that I think that rsep and smsep aren't necessarily redundant in the case where you want to oversample a received signal.

It looks like the tsc and tus variables are marking the time when a sequence starts (again, OS Working Group is probably better source) as the computational overhead of copying one sequence to memory may take more than one sample.  The time.* variables mark the beginning of the beam dwell time.

There are many of the _informational_ fields that you should not remove as they are used by plotting software as well as record keeping(data managers).

asreimer, 2018-08-09:
Hey @brianbienvenu,

TL:DR;
1) Some parameters are old and probably no longer needed (leftover from analog receiver days)
2) Follow @ksterne's suggestion and contact OS Working Group
3) IQDAT -> RAWACF might not "just work"<sup>TM</sup> when using `make_raw`

The IQDAT file format stuff that you found in my thesis should be identical to what @egthomas linked you to. When I started my thesis, this documentation wasn't so easy to find, so I included it as an appendix.

I just wanted to partially echo what @ksterne said, but also push back a tiny bit. There are parameters in the IQDAT files that __were__ there for a reason, so I would not necessarily remove any of them, but the reason why some of those fields are there might be knowledge that needs to be rediscovered by talking to very specific people. It could also be that whatever the reason was, it is no longer valid and the fields are no longer needed. I'm thinking about some of the variables like `stat.agc` and `stat.lopwr`, which seem to be from back in the good ol' days of analog receivers. Other variables are completely baffling (`mxpwr` has units of kHz? What?).

Please follow @ksterne's suggestion to talk to the OS Working Group. There are definitely metadata type fields that are **definitely not** redundant. smsep and rsep are a good example. Another good example is ltab and ptab. Consider the 8 pulse sequence that is usually used, katscan, which has 28 possible lags, which you can get from the ptab, but you'll notice that the ltab doesn't include all 28 lags, for example, lags 42 and 43 are missing for example.

@ksterne, where does one find out who is on the OS Working Group, and how to contact them? That could be helpful information if you have it handy.

@kevinkrieger @kkotyk @mts299, how are you guys handling this problem with the Borealis project? You will have to make IQDAT files too, so how are you filling in these parameters?

A heads up on using `make_raw`: I've processed IQDAT files from the SuperDARN Canada radars and had to make modifications to the `make_raw` that is shipped with RST. If you really start digging and try to process IQDAT files from different radars that are run by different groups, you'll find that `make_raw` shipped with RST doesn't work for some of them. Just a heads up on what to expect. That being said, the modifications to get things working should be minimal. You can see here what I had to do to get `make_raw` working with sas radar IQDAT files: [my edits](https://github.com/asreimer/RSTLite/blob/b452a70b5045d308e0f555768dabc518a57ce7ed/codebase/superdarn/src.bin/tk/tool/make_raw.1.9/make_raw.c) and [the make_raw currently in rst](https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.bin/tk/tool/make_raw.1.8/make_raw.c).

Here's the diff for your convenience: [diff.txt](https://github.com/SuperDARN/rst/files/2272483/diff.txt) The biggest difference is that I had to include a couple new command line switches so that I could manually tell `make_raw` how to handle SuperDARN Canada IQDAT files. After asking questions for 5 years, I found out that this is because the IQDAT file format was never formally agreed to in the PIs agreement, so different SuperDARN groups do slightly different things with how they use the IQDAT parameters. For example, the SuperDARN Canada decided to reuse the __chnum__ parameter, which was originally used to specify whether a file had STEREO data in it or not, to specify whether or note interferometer array voltage samples were collected.

I'm mostly telling you all of this because you are at the stage in the game where you can make design choices that allow your IQDAT files to just work with `make_raw` in rst. 

Once you have a RAWACF file, everything else works great. You can just use RST to get fitacf files. The trickiest bit is generating IQDAT files and then IQDAT -> RAWACF.

pasha-ponomarenko, 2018-08-09:
The first name is Mike and you might guess the surname...
;-)

ecbland, 2018-08-09:
@asreimer 

> There are definitely metadata type fields that are definitely not redundant. smsep and rsep are a good example. Another good example is ltab and ptab. Consider the 8 pulse sequence that is usually [...]

Doesn't the `mplgs` parameter take care of this? In katscan we have mplgs=23, so the software knows to exclude the last few lags.

In FITACF3.0 there is an option (in the form of commented-out code) to calculate the lag table (`ltab`) directly from the pulse table (`ptab`). I'd be interested to know why this isn't the standard way to do it. 

Since `ltab` is hard-coded in the control program, mistakes are possible (and do exist - contact me for more info).

brianbienvenu, 2018-08-09:
Thanks for all the feedback! Noted that `smsep` and `rsep` are distinct.

I wasn't sure if the OSWG was still active - I couldn't find any recent meeting notes or annual reports and their [list server](http://lists.le.ac.uk/mailman/listinfo/darn-oswg) seems to be down for web traffic.

An email to the DARN-oswg-join@ion.le.ac.uk (listed in the OSWG Charter on the VT site) bounced back with "550 Unrouteable address". I've just emailed darn-oswg@lists.le.ac.uk directly, which hasn't bounced.

Regardless of all this, it seems like the first level of files which are commonly distributed amongst the SD community are the RAWACF files, which are standardised enough that FITACF 2.x, 3.x, LMFIT, FITEX etc all read them without modification.

If modifying `make_raw.c` to suit individual hardware is a common and accepted part of SD, I may make an alternative IQ format to suit our hardware, and then ensure that I have a `make_raw` which produces standard files. We have some unique hardware capabilities which don't fit well into the normal IQDAT format: 1) two interferometers 2) 32-bit sampling 3)

@asreimer:
>Another good example is ltab and ptab. Consider the 8 pulse sequence that is usually used, katscan, which has 28 possible lags, which you can get from the ptab, but you'll notice that the ltab doesn't include all 28 lags, for example, lags 42 and 43 are missing for example.  

@ecbland 
>In FITACF3.0 there is an option (in the form of commented-out code) to calculate the lag table (ltab) directly from the pulse table (ptab). I'd be interested to know why this isn't the standard way to do it.

Me too! On the lag-cutoff `mplgs` parameter - is there a paper/presentation describing why the longer possible lags are disregarded?

ksterne, 2018-08-09:
@brianbienvenu, you've seen my note, but in case others aren't on the darn-oswg list I cc'ed Mark on this and another e-mail that was sent to that list but more relates to this repo.  I followed up with Mark directly to see what the status is for an update to SuperDARN mailing list software.

My apologies for not including much of any helpful link on how to get ahold of the OSWG.  Some of that is not knowing where to begin or point you to, especially with the e-mail problems.

From @asreimer:
> Some parameters are old and probably no longer needed (leftover from analog receiver days)

Some of the parameters may no longer be used but could be kept there for backwards compatibility.

I think I've passed this along that while I see there's a `make_raw` function, I'll second Ashton's caution on using it straight out of the box.  I'm not sure many (if any) use this binary.

> If modifying make_raw.c to suit individual hardware is a common and accepted part of SD, I may make an alternative IQ format to suit our hardware, and then ensure that I have a make_raw which produces standard files.

While iqdat files aren't the shared data product, I would discourage this kind of approach as it makes collaborating across different institutions (or even different people within a single institution) much more difficult when you have special software to read/analyze data.  I understand the path forward then is having the OSWG hold the definition for each data product that is produced by radars, but as noted before, having that happen seems slim-to-none considering the inactivity and e-mail list troubles.  This is something I called for the DAWG to take up, but we haven't gotten much of anywhere outside of what already exists (link that @egthomas initially provided).

asreimer, 2018-08-09:
@ecbland 
>Doesn't the mplgs parameter take care of this? In katscan we have mplgs=23, so the software knows to exclude the last few lags.

That's only true to a point. You are assuming things about your pulse sequence that may prevent you from supporting custom experiments. Things like the ltab is in order and that there the only pulse sequences that will ever be run are the standard  7 pulse Schiffler and 8 pulse Katscan sequences. What about custom experiments with completely different pulse sequences? Or sequences that have non-redundant lags, such as tauscan? Having both a `ptab` and `ltab` enables flexibility. No one has yet shown that the pulse sequences we use are actually optimal for making the measurements we want to make. This is true for both the SuperDARN and ISR communities actually.

@brianbienvenu 
> If modifying make_raw.c to suit individual hardware is a common and accepted part of SD, I may make an alternative IQ format to suit our hardware, and then ensure that I have a make_raw which [...]

That may be what you have to do, but I would encourage you not to for the same reason that @ksterne mentioned. Also I disagree with your interpretation of what I wrote. The IQDAT format is still respected to a larger extent than you have understood. Don't forget that the changes I had to make to `make_raw` are very small. Yes, you have some different additional things, but that doesn't mean the IQDAT format couldn't be extended in some way to include these changes. TIGER-3 isn't the only digital radar in development. I would encourage you to talk with [Bill Bristow at UAF](http://cem.uaf.edu/ece/people.aspx) and [Kathryn McWilliams at the University of Saskatchewan](https://artsandscience.usask.ca/profile/KMcWilliams#/profile) (am I missing any others? AGILEDARN?).

This would be a wasted opportunity to extend the IQDAT file format within the community. And you are going to need to potentially have these conversations anyway when you start making the RAWACFs. AFAIK, the RAWACF files are a PI agreed to standard. How does the RAWACF standard work for multiple interferometer arrays?

As you can see, the stick is very pointy.

Finally:
>Me too! On the lag-cutoff mplgs parameter - is there a paper/presentation describing why the longer possible lags are disregarded?

There definitely isn't a paper, but there might be a presentation somewhere (there are papers that show the pulse sequence, [such as this one](https://agupubs.onlinelibrary.wiley.com/doi/pdf/10.1002/rds.20031), just never explain why some lags are never saved). I think Kathryn maybe made one. The reason is that the longer lags don't usually provide meaningful information. If you work through all the possible lags for katscan for instance, you'll see that lags 0 though 24 only have lags 6 and 23 missing, however the remaining lags (27, 31, 42, and 43) all have huge multi-lag gaps between them. From a data fitting perspective, these gaps are bad because they have very low Nyquist velocity capabilities (useless for ACFs with high velocity) and are really only able to provide information about the decay envelope of the ACF. From a scattering physics and measurement technique perspective, depending on the pulse spacing and the ACF decorrelation time, these lags are usually nearly 0 anyway (except for ground scatter and the odd time ionospheric scatter has a long correlation time). Personally, the data analysis reasons are compelling enough, but they are related to the scattering and measurement physics.

pasha-ponomarenko, 2018-08-09:
@ecbland 

> In FITACF3.0 there is an option (in the form of commented-out code) to calculate the lag table (ltab) directly from the pulse table (ptab). I'd be interested to know why this isn't the standard way to do it.

I might try to guess why this was done in the past. The main SD data product is the LoS velocity, and its accurate determination depends on the accuracy of the ACF phase "unwrapping". The latter requires contiguous lag sampling so that the pulse sequences were selected based on their ability to generate a contiguous set of lags starting from lag0. I assume that the larger lags beyond the contiguous range (i.e. adjacent to gaps) were considered to be of little importance because they cannot be used for two-pi unwrapping. This consideration combined with the need to save disk space (it was important 25 years ago) was most probably the main reason for not including all possible lags in the ACF calculation.
The commented out code for determining all possible lags has been developed by Keith at the very beginning of his work on FITACF3: he did not realise at first that not all lags are used.
Personally I think that we have to use all available data, but it needs to be investigated by how much those extra lags would improve the data quality.

pasha-ponomarenko, 2018-08-09:
I see that Ashton has already commented along the same lines...
;-)

ecbland, 2018-08-10:
@asreimer @pasha-ponomarenko  Thanks for the detailed response. That makes sense regarding sequences such as tauscan.

closed 2019-04-10 by egthomas