SuperDARN/rst #178: PC installation - extra environment variables to define

arfogg, 2018-08-10:
Hi,

When installing RST 4.1 feature/ts18model I had to define some extra environment variables to get the  code to compile correctly, other than the stuff pointed out in the installation notes. I had the same issue when installing RST 4.1 a while back, but @egthomas helped me to comment out some of the CDF stuff as I don't use it. It could be handy to have these variables also noted as something you might need to change.

Here are the extra environment variables I had to edit:
  in rst/.profile.bash
export PATH="${PATH}:/usr/bin/:${HOME}/bin:${RSTPATH}/bin"

  in rst/.profile/idl.bash
export IDL_PATH="${IDL_DIR}:${RSTPATH}/idl/lib"
export IDL_DLM_PATH="${IDL_DIR}:${RSTPATH}/dlm"

  in rst/.profile/rst.bash
export LD_LIBRARY_PATH="${LIBPATH}:${USR_LIBPATH}:/data/sol-ionosphere/cdf/cdf36_3-dist/lib"

  in rst/.profile/superdarn.bash
export ISTP_PATH="/data/sol-ionosphere/istp"
export SD_HDWPATH="/data/sol-ionosphere/rst_4.1-ts18/tables/superdarn/hdw/"
export SD_RADAR="/data/sol-ionosphere/rst_4.1-ts18/tables/superdarn/radar.dat"

( for the last two it's just a hard coded location for RSTPATH, so that might not be as necessary, but it's something I did to get the compilation to work )

Redefining these variables was necessary to get it to work on our system - this could be due to the way our system works, but I hope it's helpful to note this in case anyone else is having similar issues.

It's all installed now and works fine, but in theory I could replicate the problem if necessary.

Thanks,
Alexandra

asreimer, 2019-01-27:
Hey @arfogg, can you give me some more information about how you were installing RST here?

What OS? You say PC, so doesn't this mean Windows? I'm interested in testing this and helping develop documentation to help other users get RST installed.

mts299, 2019-04-02:
@arfogg Did you resolve this issue? 
Can we close it now?

arfogg, 2019-04-05:
Hi,
Apologies for the late reply.

I installed RST onto our NoMachine / Linux system - apologies for the confusion from me describing it as PC. 

RST installed successfully and works fine for me, just wanted to comment that I had to change a few extra environment variables than were mentioned in the install notes.

Thanks,
Alexandra

closed 2019-04-10 by egthomas