SuperDARN/rst #180: RSt testing folder

mts299, 2018-08-13:
While working on the elevation merger between fitacf 2.5 and 3.0, I ran into some issues with fitcomp.1.0 in /superdarn/src.bin/tk/testing/ folder.
Somehow, when I include the libraries I need for FitACF to work with my new scheme the fitacfex library could not be built. After talking to @asreimer, it seems this file only tests various fit algorithms, and it does it by commenting and uncommenting code. @asreimer further looked into the testing directory and realized this is an AJ thing for testing code for a paper. 

My main issue is should we delete this folder? Seems to add no value and it may cause more headaches than anything.

asreimer, 2018-08-13:
@ajribeiro, can you provide a bit of info here? I think you originally authored some of the contents of [superdarn/src.bin/tk/testing/](https://github.com/SuperDARN/rst/tree/master/codebase/superdarn/src.bin/tk/testing)?

I think this was done for the [comparison of fitting methods paper](https://agupubs.onlinelibrary.wiley.com/doi/pdf/10.1002/rds.20031).

ajribeiro, 2018-08-13:
Deleting it should be fine.  If for any reason the code is needed in the future, it will still be available in an old version of the repo.

egthomas, 2019-06-19:
Closing this now that the `fitcomp` folder has been removed in #93.

closed 2019-06-19 by egthomas