SuperDARN/rst #181: potential field_plot issue

mts299, 2018-08-23:
Hey, 
Guowei Wang is having problems with RST 4.1 field_plot saying that he gets an error when trying to read in the files:
20160616 han or zho fitacf 
says the error is "error: reading file"

I suggested to him to try RST 4.2 (sent him a tar since they have no access to github) and that I would also try on my end since the time zone difference makes us work in different times of the day. With that said our mirror is down and we don't have the files on our server machines... yay... 

So I am wondering if anyone could test this out for me in RST 4.2 release please? @egthomas @ksterne ? 

My suspicion is it might be scan flag = -1 or he is using bad files :( 

Note: these files are not blacklisted. 

Thanks!

egthomas, 2018-08-24:
Both radars were using the stereoscan control program and have negative scan flags on that day; you're correct that this was fixed by #132 in v4.2.

mts299, 2018-09-05:
RST 4.2 fixed his problem!

closed 2018-09-05 by mts299