SuperDARN/rst #184: Convection Map procedure

mts299, 2018-09-05:
Another question I received from Guowei Wang is the process of producing a convection map. 

Are there any clear instructions I can send him on the steps to making a convection plot?

egthomas, 2018-09-05:
This is a slightly outdated document (ie some options may no longer exist or have since been changed) but is still probably the most comprehensive set of instructions available:

http://superdarn.thayer.dartmouth.edu/documentation/manual.pdf

mts299, 2018-09-05:
Okay, I will pass it on to him and maybe my scripts I use to generate the convection plots so hopefully that will be fine for now before we have a RST for dummies manual

mts299, 2018-11-08:
I am going to close this issue for now. Issue #195 is addressing a better document for teaching users how to use RST and I have not heard back from Guowei Wang.

closed 2018-11-08 by mts299