SuperDARN/rst #185: Goose bay elevation infinity bug

mts299, 2018-09-18:
@egthomas found a bug in the goose bay elevation algorithm that seems to generate an infinite loop when calculating the `dchi` value when testing the PR #150 

> Something needs to happen with this - I'm seeing (seemingly) infinite changing dchi by -2pi. ... -> ... output when trying to process 1994111000g.dat with make_fit in RST 4.2 at or after record 1994-11-10 0:41:25 beam=15. Same thing happens whether it's the original dat file or converted to rawacf. 

> It's originating from this iterative code block in elev_goose.c:
https://github.com/SuperDARN/rst/blob/master/codebase/superdarn/src.lib/tk/fitacf.2.5/src/elev_goose.c#L125-L138

@pasha-ponomarenko also mentioned that the values are very large for `dchi` 
> changing dchi by -2pi. -595562.236100 -> -595568.519286

pasha-ponomarenko, 2018-09-18:
@egthomas, did you try RST4.1? 
Also, do you know if there is at least some description of what Kile implemented in elev_goose.c?

egthomas, 2018-09-19:
@pasha-ponomarenko the `fitacf.2.5` library did not change between 4.1 and 4.2, and the only change to `make_fit` was the addition of the `--version` option.

mts299, 2018-09-19:
@egthomas if you were just testing on my branch please pull, as I added a fix for fitacf 2.5 where I was not setting `bmnum` in my elevation structure. 

Please test now :)

egthomas, 2018-09-19:
No I'm not doing any testing, I've just been using the v4.2-tagged archive for data processing.

mts299, 2018-09-19:
Okay, so this is not a bug I potentially caused with my new implementation?

egthomas, 2018-09-20:
No, this issue was already present in the software.

mts299, 2018-09-20:
Okay, I will keep my coding eyes open if anything comes up when I am testing goose bay for my branch.

egthomas, 2018-10-24:
Going through some older emails, Gareth Chisham and Rob Shore at BAS actually encountered this issue and brought it to our attention back in January.  At the time I didn't appreciate just how much of a problem it was when trying to mass-process Goose Bay data.

sshepherd, 2019-06-18:
As much as it pains me to comment on an algorithm that has been demonstrated (several years ago) to be, at best, an approximation, the cause of the infinite loop behavior has to do with the error associated with the phase (I'll call it phi0_err here). In nearly all the cases where the algorithm gets stuck, the value of phi0_err is 327.670, which is in radians. There are a couple of exceptions in the file that Evan mentioned recently (1994111000g.dat), but almost all of the problem records have this large value for an error.

In do_fit(), elev_goose() is called three times in succession to calculate the normal, low and high values of the elevation angle using phi0, phi0+phi0_err, phi0-phi0_err. The low value seems to be the culprit since the starting value is pushed well beyond the maximum phase possible and the loop continues to iterate further away from the maximum value. The high value is not great, but does converge in ~50 iterations.

I am unclear as to why there are large values in phi0_err. It seems that they should be limited to +/- pi. Perhaps a better solution is to trap values of phi0_err that exceed abs(pi) and just set the low and high values to something silly, say +/- 999? Then again, I'm not sure why we are still using this algorithm.

ecbland, 2019-07-23:
Any objections to closing this issue? 
1. @egthomas implemented a loop counter as a band-aid fix for this bug (#217). 
2. @sshepherd's new elevation angle calculations have been merged, so the `elev_goose` code is only called with the `-old_elev` option. I doubt that many users will want to use the old code when we have known for several years that the calculation is incorrect. 

If there is motivation to investigate this bug more thoroughly then we can keep this issue open, but I think we all have other priorities.

ksterne, 2019-07-25:
I usually like to make sure the originiator of the issue is satisfied, so @mts299, seem good?  I think @sshepherd might also have a weigh in here now that #93 has been merged in.

mts299, 2019-07-25:
Huh forgot this issue was even still open, I thought it was closed with the merging of #217. 

I am going to close it with a workaround label just to be clear we never really found the bug but that was due to time effort constraints.

closed 2019-07-25 by mts299