SuperDARN/rst #195: RST for Dummies

mts299, 2018-10-30:
This issue is not much of an **issue** as it is a start of a discussion of a user manual. Recently I have been helping people with creating convection plots in RST as it is not very clear where to find the information and what the process involves. 

So I am wondering if we could start something now (maybe a google file) as a rough draft? 
We could just throw topics in, and little blurbs of what we say to people when explaining things about RST to remind us what people have troubles with and what we said that we found helped them.

pasha-ponomarenko, 2018-10-30:
@ecbland, since you've already started writing...

mts299, 2018-10-30:
@ecbland, If you could do that would be great :+1:

ecbland, 2018-10-31:
Ok, here's what I have so far on RST plotting routines, dumped into a google document:
https://docs.google.com/document/d/1EUyTexThamKLAkR_6Wlhp4GcASSmAJSwJJTML-edqUk/edit?usp=sharing

Anyone with the link can view and comment. If you would like to edit, please contact me directly. 

You'll notice that there isn't anything yet about plotting convection maps; I will add them soon, unless someone beats me to it! I also have notes on making FIT/GRD/MAP files, and some other random things, which I'll add to the document when ready.

mts299, 2018-11-01:
@ecbland Thanks for doing this!

pasha-ponomarenko, 2018-11-06:
@ecbland, I have just compiled a short description of the main fitted parameters, which should be useful here:
[Main fitted parameters.docx](https://github.com/SuperDARN/rst/files/2551554/Main.fitted.parameters.docx)

Please feel free to edit/correct it.

pasha-ponomarenko, 2018-11-06:
Already found a typo:

> Spectral width IS used

billetd, 2018-11-08:
I've recently had the same experience with a new student @mts299, it's definitely not the clearest. @ecbland We have a fairly old manual from an RST once past which has proven quite useful to get people started, though most of the flags/commands are pretty outdated.
[RST_manual.pdf](https://github.com/SuperDARN/rst/files/2561909/RST_manual.pdf)

ksterne, 2019-01-11:
Is it possible that this link could be more visible?  Maybe an addition to the main README.md file or to the Data Analysis WG website?

pasha-ponomarenko, 2019-01-11:
Added to DA-WG page during my tea-break from the IT duties...

aburrell, 2019-01-14:
A link on the README.md file would also be a good idea.  That's more most people (especially those who don't know about the WGs) will think to look.

ecbland, 2019-01-14:
I agree that a more visible link and inclusion in the README file is a good idea. However, I'm not happy with the google documents format (e.g. it's hard to include code snippets, the figures don't always stay where you want them, and the layout is not friendly). I'd like to investigate some alternatives, so could we hold off on this for a few days while I do this? Or am I the only person who feels this way?

aburrell, 2019-01-14:
No, @ecbland I agree with you.  But I also think that what you have now is much better than what we had previously so am willing to settle for incremental progress.  However, a few days won't matter.  @mts299 should this be reopened until we have a better document and a link in the README?

egthomas, 2019-01-14:
Would it be difficult to port the existing guide to latex?  I'm not sure about the figures, but at least the source text could then be stored in a github repository.

ecbland, 2019-01-14:
Converting it to LaTeX is definitely an option. 

We have been using Dokuwiki at UNIS for our internal SuperDARN stuff - this is where I originally wrote the stuff that's now in the google document. We have been really happy with it: easy to use and it works well with multiple contributors. Just need to sort out how to host it. 

Other suggestions welcome, and I will continue to investigate