SuperDARN/rst #203: OldFitOpen does not read minor FITACF version numbers correctly

egthomas, 2018-12-04:
As the title indicates, minor FITACF version numbers are not correctly handled when working with the old `.fit` format.  For example: 

1. Change the `FITACF_MINOR_REVISION` value in [fitacfversion.h](https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fitacfversion.h) from 0 to 5 (as we've done on a few branches such as `elevation_library`)
2. Create a `.fit` file from a `.dat` file
3. Perform a `fittofitacf` transformation to produce a `.fitacf` format file
4. Finally, use `dmapdump` to view the contents of the `.fitacf` file; the `int "fitacf.revision.minor"` field seems to always be equal to 0 (ie not 5 as we specified in `fitacfversion.h`)

I believe the offending code is located somewhere in this section of `OldFitOpenFd` in `fitopen.c` in the `oldfit.1.25` library:

https://github.com/SuperDARN/rst/blob/develop/codebase/superdarn/src.lib/tk/oldfit.1.25/src/fitopen.c#L106-L116

mts299, 2019-01-18:
@egthomas did you fix this in your PR #206? If not can you please provide some info on what also needs to be done.  You can also use github's checklist ability (then someone can just check it off as they make a PR for it :+1: ):

- [ ] Change the FITACF_MINOR_REVISION value in fitacfversion.h from 0 to 5
- [ ]  Create a .fit file from a .dat file
- [ ]  Perform a fittofitacf transformation to produce a .fitacf format file
- [ ]  Finally, use dmapdump to view the contents of the .fitacf file; the int "fitacf.revision.minor" field seems to always be equal to 0 (ie not 5 as we specified in fitacfversion.h)

egthomas, 2019-01-18:
Yes, I believe this is fixed in #206 (where it looks like you did a very thorough job of testing!).  Thanks for the suggestion about the checkboxes, I'll try to remember to use those moving forward.

mts299, 2019-01-18:
Closing the issue then since PR #206 addresses it.

closed 2019-01-18 by mts299