SuperDARN/rst #205: cnvmap format documentation needs updating

ksterne, 2018-12-06:
Seems as though we should update the cnvmap variable definition documentation for the lastest variables that were added in around release 4.2.  Here the documentation is found in: `~/doc/html/superdarn/rfc/0018.html`.

Also opening up the `index.html` file in `~/doc/html/` and clicking through "superdarn -> General -> rfc" leads me to a blank page and not the listing of file formats and other things.  Could this be an issue of looking at it through local files (file:///) and not a web connection(http(s)://)?  Does anyone else have this issue or just local to my virtual machine?

egthomas, 2018-12-07:
This is along the same lines as the issue @kkotyk raised in #194.  I returned the RFC docs to the RST because they had useful information, but unfortunately some/most of it has become outdated.  The decision needs to be made whether to keep them "as-is" for historical reference, or try to update and maintain them moving forward.  It may be worth noting the RFC docs were removed from the RST sometime between versions 2.x and 3.x - I don't know if they were updated or maintained independently somewhere else on the old APL site.

@ksterne regarding your compilation issue, you may need to execute the `make.doc.rfc` script in the `build/scripts` directory:

https://github.com/SuperDARN/rst/tree/master/build/script

ksterne, 2019-01-17:
I would think making updates would be preferable.  Though does it make sense to keep these as RFCs or at some point label things as documentation/reference?  This should be fairly simple to update, and I could say I could get to it, but I'm massively behind on documenting/writing other things for the moment.

mts299, 2019-01-18:
Sorry if I am asking too many questions but I feel this was happening before my employment. 
@egthomas were the RFC's (assuming just html pages) stored on github? If so then any changes made to them should stay in the commit history so we shouldn't be afraid of making any changes to them. If anything they should follow changes to the code so releases are kept consistent with code/documentation changes.

egthomas, 2019-01-18:
@mts299 the RFCs were copied by me from an older version of RST (I think 2.11 - https://github.com/SuperDARN/rst-archive/tree/rst.2.11/code/rfc) and then I made whatever necessary changes (possibly none) to get them to compile with the current era of software.  You can see from the link on @pasha-ponomarenko's DAWG website how the RFCs used to appear on the old APL website:

https://homepage.usask.ca/~pbp672/Data%20Analysis/fit_data_format.pdf

I don't know what decisions lead to the RFCs eventually being removed from the RST with the 3.x era of software.

egthomas, 2019-01-18:
Looking back, we had a very similar discussion nearly 2 years ago in the original pull request (https://github.com/SuperDARN/rst/pull/47) with no consensus reached.

mts299, 2019-01-18:
Okay, I raised a new issue discussing RFC's future #210 since I think this issue is focused on cnvmap format of the html page. 

@ksterne did the `make.doc.rfc` fix your original issue?

ksterne, 2019-01-22:
@mts299, that's a good new issue you've started which addresses the topic of the format of our documentation.  I'd like to keep this open as the original topic is they content of the documentation needs to be updated.

Also, yes I think `make.doc.rfc` fixed the problem and things worked from there.

mts299, 2019-01-22:
@ksterne who could update the documentation?

closed 2019-04-12 by egthomas