SuperDARN/rst #207: Folder not found Segfualt

mts299, 2019-01-11:
While testing PR #204, I noticed when I used the `-path` option with `map_plot` I got a `segfault` if the directory did not exist. 

Possible solutions:
1. create the directory if it does not exist 
2. Print an error that the "Folder does not exist"

Otherwise, this segfault is miss leading about what is wrong and poor practice.

egthomas, 2019-01-11:
Nice catch.  My vote would be to print an error and exit rather than try to create a directory (which could run into other permission issues).

mts299, 2019-04-02:
@egthomas This was resolved in #204 correct? 
If so then we should close the issue :+1:

egthomas, 2019-04-02:
@mts299 no, I don't think that pull request modified any of the plotting code.

ksterne, 2020-02-20:
@mts299, I know you're out here so I'll close this since it was solved in #262.

closed 2020-02-20 by ksterne