SuperDARN/rst #210: RST Documentation Revamp

mts299, 2019-01-18:
I have noticed a lot of "issues" #201 #197 #195 revolving around documentation lately which I think is amazing :+1: 
However, the question might be where to put things and how to move forward in our documentation.
The following discussion points:

- [ ] RFC's are they something to keep maintaining or should we archive them and move to something different? 
- [ ] More user-friendly documentation setup for RST functionalities? Read the Docs might be an option. 
- [ ] RST for dummies - where should the link live?
- [ ] RST for dummies - what platform should it be shared on Overleaf vs. google doc? 

I raise these questions to give us more of a direction on documentation, also I find the current sifting through RST documentation annoying and I will be starting to document pydarn with Read the Docs and I am willing to start an RST version as well to kill 2 birds with one stone. 

Also, RST for Dummies can get discussed here to help figure out its home and other details. 

Let's discuss!

aburrell, 2019-01-18:
I think Read the Docs is a good place for the RST code documentation.  Even if we decide to move it in 10 years or so, we'll have all the documentation written and in an easy-to-read format.

mts299, 2019-01-18:
I have no problem with starting to transfer information over from RFC's to read the docs as long as there is potential future for it :)

ecbland, 2019-01-18:
I don't have any experience with Read the Docs, but would it be an option for hosting RST for Dummies?

aburrell, 2019-01-18:
Yes, though perhaps not the best option.  It has some good features for coding examples and document strings, but (as far as I know) is not as good for creating a parallel pdf that can be downloaded and digested offline.  But other people may be more experienced in its use and have other opinions.

mts299, 2019-01-18:
@aburrell, you beat me to the punch. Read the Docs would be good for code documentation. 

Would markdown or latex be of interest for RST for Dummies? Markdown we can host in GitHub but also make it pdf available.

pasha-ponomarenko, 2019-01-18:
A note in passing: I am not sure about _markdown_, but I like the old good _LaTeX_: it is easy to use, and, more importantly, **I** know how to use it!
;-)

egthomas, 2019-01-21:
It's also worth remembering that while the current .xml documentation is a pain to sift through in its html format, it is also used to build the command-line help which some people use quite frequently.

mts299, 2019-01-21:
@egthomas can you elaborate on the process of using XML to build the command-line help?

asreimer, 2019-01-22:
+1 to moving towards a documentation format that is both inline with the source code and automatically updated using tools like readthedocs.

ecbland, 2019-01-24:
Quick update on RST for Dummies: I am in the process of converting the google document to LaTeX. I'll try to have it done by next week.

ecbland, 2019-02-25:
Finally got around to putting the RST for Dummies into a LaTeX file: [RST_user_guide.pdf](https://github.com/SuperDARN/rst/files/2902613/RST_user_guide.pdf)
It's not finished, but this should be enough to get an idea of scope, layout etc. Comments welcome. 

Question: is this printable PDF format preferable to putting this content on the wiki? Perhaps #215 is relevant here.

aburrell, 2019-02-26:
As long as we can all edit the LaTeX file, I think that's fine.

kkotyk, 2019-02-27:
@aburrell RTD does have a feature to download an RTD project as a pdf. if you go to a page and then click the little green text with the arrow, you can choose to download as pdf or HTML. 

RTD can also generate documentation from LaTEX too.

kkotyk, 2019-02-27:
LaTeX snippets can also be inserted into RTD's restructured text markdown as a FYI.

asreimer, 2019-02-27:
This is awesome @ecbland!

Is there any pushback against putting all of this up on RTDs? RST should probably have some kind of centralized documentation and it is a pretty common practice for software projects on github to host this in a /docs directory and have RTDs build from there. The documentation can then be updated by contributors as part of PRs for features/bugfixes.

mts299, 2019-04-10:
Before we start the meeting just wanted to provide a link on something I have been working on in my spare time: https://radar-software-toolkit-rst.readthedocs.io/en/latest/
It is the start of the Read The Docs for RST and I am looking at combining it with doxygen for package structure documentation.

mts299, 2019-07-26:
Since we have started using ReadTheDocs in PR #246 can we close this issue when it is merged and open up any other issues with documentation down the road which then we can decide if it stays with the old or goes into the readthedocs?

ecbland, 2019-07-31:
@mts299 Closing this issue now since we seem to have finished the discussion. We can open new issues on the documentation as they arise.

closed 2019-07-31 by ecbland