SuperDARN/rst #212: IMF missing data

mts299, 2019-01-22:
@egthomas brought up this point in PR #204 

> The question of how best to indicate missing values (ie 0 vs 9999) is probably best left for a later discussion.

@egthomas can you please elaborate on what you mean? 
Is 9999 not an obvious missing value? 
Isn't 0 an actual possible value for the IMF to be?

egthomas, 2019-01-23:
I don't know if changes need to be made to `map_addmodel` or if at some point it falls to the user to determine an IMF value of 9999.9 is unphysical and the map potential solution should maybe not be trusted (depending on the number and extent of input vectors).

pasha-ponomarenko, 2019-01-23:
Just a quick note: generally, zero is not a good flag for "bad" data.

egthomas, 2019-01-23:
> Just a quick note: generally, zero is not a good flag for "bad" data.

@pasha-ponomarenko agreed

ksterne, 2019-01-23:
@egthomas, sorry you haven't answered what I was after so I'll ask it a different way.  In #201, I noted there were two (possible) problems.  One with `map_addimf` and the other with `map_addmodel`.  The first was fixed with #204.  So I'm wondering if this issue is the second part so that I could close out #201.

egthomas, 2019-01-23:
The second problem from #201 you're talking about is that `map_addmodel` sees an IMF value of 9999 and doesn't realize that means missing IMF data, correct?

If so, my point is that `map_addmodel` doesn't need to make a correction - the IMF values will be bogus / unknown regardless of what is used as a missing value indicator by `map_addimf`, so the map potential solution should be treated with extra caution by the user.  I don't believe that is an issue which needs to be addressed in software.

If not, apologies for the continued confusion on my end.

egthomas, 2019-01-23:
Oh sorry, you meant whether you could close #201 - my mistake.  Yes, I think this issue covers that second question so you can close #201.

mts299, 2019-01-23:
@ksterne Sorry if this ended up being a duplicate of issue #201 I was not where labeling "bad" data fit into everything. 

@egthomas doesn't `map_addimf` replace missing values (indicated by 9999.9 (I think it is possible to think these are unphysical values and just fillers since 0 is a possibility.)) now so `map_addmodel` would not see 9999.9? 

Moving forward, an idea for dealing with missing data could be handled alone in `map_addimf` where missing data algorithms are implemented like the statistical models used in convection maps that user can choose from. That way we provide some aid to help the user, but it is up to them to be aware of what they are using and how it may affect other parts of the convection map process.

mts299, 2019-04-02:
Closing this issue. Maybe we can review it again when some one has time to implement various missing data methods.

closed 2019-04-02 by mts299