SuperDARN/rst #213: Segfault on file not found for plotting commands

mts299, 2019-01-23:
While testing #209 I found that you get a `segfault` when the file is not found:

`time_plot -png -gs -v -b 5 > time_2_5_sas.png`
`Segmentation fault (core dumped)`

I assume this could be a quick fix since we have done it for other binaries like make_fit.

egthomas, 2019-01-23:
We probably just need to add the same input file check that we did for the `map_plot` and `grid_plot`: 
 https://github.com/SuperDARN/rst/commit/e373baae000ea6581ba0c421e86e6f0db39c740f

egthomas, 2019-01-23:
It turns out that was a typo in `time_plot.c` - for whatever reason that plotting function can accept stdin as input.  Please take a look at #209 and let me know if that is an acceptable fix.

egthomas, 2019-06-19:
Closing this now that the fix in #209 has been merged.

closed 2019-06-19 by egthomas