SuperDARN/rst #222: DMAP file checking

mts299, 2019-02-27:
Lately, there has been more file checking going with PR #221 and #139; I think this might be the time to review the idea of file checking in RST. I understand that DDWG should handle most of the blacklisted files; however, our system is not perfect. @kkotyk brought up the point that this can be a touchy subject in PR #221 and maybe a further discussion is needed, or we are at the compliance that we are going to do it anyway so might as well do it now :p 
 
Side note, I was going to make a branch that does this and make the PR the discussion area to see what effect having file checking implemented can bring to the table for RST. Also, if people were against the idea of merging, it could be there as a reference or use for those that want to use file checking ¯\_(ツ)_/¯

asreimer, 2019-02-27:
A summary from my perspective having worked with both SuperDARN and AMISR 100+ TB databases of radar data and performing data processing for both:

I can think of a couple reasons to have file checking built in to the dmap library:
1) Stability: no more segfaults due to corruptions in dmap files,
2) Verbosity: optionally, a user (research engineer or researcher alike) can know if there is a problem with the file they are trying to read and what that problem is. This could easily be a configurable option.

I believe these reasons were significant contributing factors, among others, for why @kkotyk built [backscatter](https://github.com/SuperDARNCanada/backscatter).

Based on previous discussions in this repo, file checking seems to fall in to 2 separate topics:
1) Handling file corruptions
2) Being verbose or not about those corruptions

Failing silently (handling a corruption but not saying anything to the user) is generally a bad practice, but could be useful in some specific situations, such as a real time display. It is likely that the vast majority of use cases for RST are post processing SuperDARN data and performing research with SuperDARN data, and in these use cases it is desirable to know if there is a problem with the data files being worked on (@ksterne and @mtwalach have discussed this).

Concerns have been raised about the additional complexity and performance hit such changes would have on RST. These concerns are mitigated in the following ways:
1) These file checks are already being added as more and more segfaults happen and get fixed, and
2) Performance won't suffer if verbosity is optional

mts299, 2019-06-13:
At the SuperDARN workshop 2019, the Data Analysis Working Group (DAWG) decide to push forward on error checking in files and throughout the RST software as we have been doing lately while keeping an eye on the performance of the code. 

Due to the time, it would take to effectively do a good job on error checking various corrupt SuperDARN files, the group decided to create error checks as things come up or when people have time to implement them.

closed 2019-06-13 by mts299